revtel/react-native-nfc-manager #341: Writing and reading NDEF records inside Iso15693 tag specific areas

ajacquierbret, 2020-11-03:
Hi @whitedogg13 !

First, congrats for your amazing work on this library !

Let's explain a bit the issue :
I need to write NDEF formated Wifi credentials inside an Iso15693 NFC tag. But this tag is splitted in four areas of 2048 bytes, and i can't figure how to write these credentials only inside area n°3 (from 4096 to 6143) for instance.

I tried using Iso15693 methods but it seems that it's impossible to provide NDEF to the `dataBlock` property of `writeSingleBlock` or even using `transceive` for Android.

Here's what I tried :

```tsx
if (Platform.OS === 'ios') {
        let tech = NfcTech.Iso15693IOS;
        let resp = await NfcManager.requestTechnology(tech, {
          alertMessage: 'Ready to scan tag',
        });
        console.warn(resp);

        let tag = await NfcManager.getTag();
        console.warn(tag);

        const handler = NfcManager.getIso15693HandlerIOS();

        resp = (await handler.getSystemInfo(
          Nfc15693RequestFlagIOS.HighDataRate,
        )) as any;
        console.warn(resp);

        const bytes = buildWifiPayload({
          ssid: credentials.ssid!,
          networkKey: credentials.password!,
        });

        const beginWifiArea = 0x1000;

        await handler.writeSingleBlock({
          flags: Nfc15693RequestFlagIOS.HighDataRate,
          blockNumber: beginWifiArea / 4,
          dataBlock: bytes,
        });
} else if (Platform.OS === 'android') {
        let tech = NfcTech.NfcV;
        let resp = await NfcManager.requestTechnology(tech, {
          alertMessage: 'Ready to scan tag',
        });
        console.warn(resp);

        let tag = await NfcManager.getTag();
        console.warn(tag);

        const bytes = buildWifiPayload({
          ssid: credentials.ssid!,
          networkKey: credentials.password!,
        });

        await NfcManager.transceive(bytes);
}
```

The simplest and cleanest solution for this use case would be to use cross-platform NDEF methods such as `writeNdefMessage(bytes)` like this :

```tsx
const bytes = buildWifiPayload({
  ssid: credentials.ssid!,
  networkKey: credentials.password!,
});

await writeNdefMessage(bytes, {
  blockNumber: 4096,
})
```

I'm really not an expert concerning NFC, and this "area" concept may be specific to the ST25DV-I2C NFC Tag, I honestly don't know, and as a front-end dev I'm a bit lost.

If you have suggestions, even platform-specific and low-level ones, it would be hugely appreciated :)

YusufBesim, 2020-11-04:
Hello, in ios 14.1 cant write nfc. Can you ? if yes how ?

lukebars, 2021-02-02:
@ajacquierbret Hey, did you find out how to use NfcV with android? iOS works fine, but transceive on android always fails...

closed 2021-05-26 by github-actions[bot]

renish-digipay, 2024-01-11:
Currently I've Nfc Tag with Technology "NfcV". and I'm working on React Native.

this is my code :  
`let tech = NfcTech.NfcV;
        let resp = await NfcManager.requestTechnology(tech, {
          alertMessage: 'Ready to scan tag',
        });

        let tag = await NfcManager.getTag();

        const bytes = Ndef.encodeMessage([Ndef.uriRecord(`${ADMIN_BASE_URL}tip/${this.state.wristbandSerialNo}`),]);
        await NfcManager.nfcVHandler.transceive(bytes)`

I'm trying now in android with this code but facing "`Error: transceive fail`".

how can i write and read the tag for both platform "Android" and "iOS". 
also i need an ultimate solution for all type of Nfc Technologies for React Native.

Thanks in Advance.