revtel/react-native-nfc-manager #342: IOS 14 Not Writing

YusufBesim, 2020-11-04:
Hello,

When we trying write ndef tag happens nothing.
Cannot continue after requestTechnology.

                  let resp = await NfcManager.requestTechnology(NfcTech.Ndef, {
                      alertMessage: 'Ready to write some NFC tags!'
                  });
                  console.warn(resp);
                  let ndef = await NfcManager.getNdefMessage();
                  console.warn(ndef);
                  let bytes = buildUrlPayload('https://test.com');
                  await NfcManager.writeNdefMessage(bytes);
                  console.warn('successfully write ndef');
                  await NfcManager.setAlertMessageIOS('I got your tag!');
                  this._cleanUp();

Thank you for help

fgrs, 2021-05-18:
same issue on ios 14.4.2

whitedogg13, 2021-05-20:
The same as #339, let's move our discussion there and close this one.

closed 2021-05-20 by whitedogg13