revtel/react-native-nfc-manager #344: Is it possible to read multiple NFC Tags at once?

jbejas, 2020-11-23:
I'm trying to read multiple NFC Tags at once, but can't seem to find a way to do so. Is there something I can do about that? Thanks!

S7860, 2021-01-24:
Same for me i have been looking for a way to read NFC tags multiple times without having to reload the app each time i want to scan the tag. Is there a way to do it please let me know thank you.

gregoryfm, 2021-04-19:
@jbejas @S7860 Any update on this?

closed 2021-07-05 by github-actions[bot]

maartenvandillen, 2023-11-09:
Yes would also like to know this. I want to be able to let the user scan multiple tags during one app session. It beeps the second time but no event is fired. Calling requestTechnology a second time produces an error with the content "Error".