revtel/react-native-nfc-manager #346: ios 14.2 iphone 8 can't read tags

IrmantasVaracinskas, 2020-11-27:
Hello,

After user initialises scan and cancels it, SessionClosed event is usually not received (sometimes it is received, I would say it works fine about 30% of the time, maybe lower) for iOS 14.2 with iPhone 8 and so user cannot read another nfc tag, also SessionClosed event sometimes is not received after successfully scanning a tag. But it works fine on iOS 14.2 iPhone XR.

Tried on 2.2.0 library version and on 1.2.6.

Haven't tried on any other iPhone.

Maybe someone has similar issue?

Thanks in advance

IrmantasVaracinskas, 2020-12-01:
Update: looks like this can also happen on iPhone XR but it is much more rare. From debugging it seems that if user calls registerTagEvent right after previous session closed then new session will be initiated but NFC scanner will not show up on screen and so user can't scan tag or cancel scanning. You have to close and reopen app.

sajadghawami, 2021-04-23:
i am experiencing that same issue!

closed 2021-07-08 by github-actions[bot]