revtel/react-native-nfc-manager #349: error: fail to connect tag

junaidrazaq, 2020-12-18:
Hi,
   I seem to be having an issue. I am trying to connect an empty tag with my app, however upon writing to the tag, i get the error: 'fail to connect tag'. This issue only persists with an empty tag. If I write to the tag using a third party app however, the tag gets written. Furthermore, when I then proceed to write using my app, the write is successful.

Upon debugging, I have found that the reason for this error is because, an empty tag contains the techtype: "android.nfc.tech.NdefFormatable". And as a result, the request technology function is not appllicable.

A tag which is not empty however, has the techType "android.nfc.tech.Ndef". This is available with the request techonology function.

I am fairly new to programming in general and would love to get some help on this issue. Is there something I could do to make the tag writable when it is empty with the techtype: "android.nfc.tech.NdefFormatable".

Thanks in advance

junaidrazaq, 2020-12-18:
So, I managed to find a fix for this.

When I tried to write to an empty tag, I would receive an error. The fix for this was to format the tag first. And so I added a function which did this. Here is the code for formatting for anyone who may need it:

///// ------ Function for formatting tag ------ \\\\\
export const formatTag = (test) => {
    test(true)
     // (erase tag)
     NfcManager.unregisterTagEvent()
     .then(() => NfcManager.registerTagEvent())
     .then(() => {
   NfcManager.requestNdefWrite(null, {format:true})
       .then(() => {
           console.log('format completed')
           test(false)
        })})
       .catch(err => {
           console.log('arrrrir', err);
         });
}

This seemed to fix it. Once I scan the tag with this function, I can then write to the tag successfully. Not sure what the issue was but glad its fixed.

avivku90, 2020-12-19:
can you explain about it a bit more please?
i get an error from your catch "Not even registered"
how and where do you implement this code? and those "test(true)" i guess are for your code use?
thanks

junaidrazaq, 2020-12-19:
So, this was the code for android. You are right, the test(true) is for my code use to open up a dialogue box.

The original code, to write to a tag is:

///// --- Function for writing tag ANDROID --- \\\\\
    export const writeTag = (type, textToWrite, wifiPswrd, test, setName, setPassword) => {

        test(true)
          NfcManager.unregisterTagEvent()
          .then(() => NfcManager.registerTagEvent())
          .then(() => {
            let bytes;
            type == 0 ? bytes = buildUrlPayload(textToWrite)
            : bytes = buildWifiPayload(textToWrite, wifiPswrd)

            NfcManager.requestNdefWrite(bytes).then(() => {
                test(false) , alert('Success! Your tag is now linked'),
            setName ? setName('') : null, setPassword ? setPassword('') : null})
            .catch(err => {
                console.log('arrrrir', err);
                test(false)
                alert('error: ZAP-D_102', err)
              });

          })
          .catch(err => {
            console.log('erro', err);
            test(false)
          });
      } 
///// ------------------------------------- \\\\\

To explain the code a little bit for you, if the type === 0, then I want to write a URL to the tag, else I want to write WiFi details to the tag. This works when the tag is not empty, however when the tag is empty, I receive an error. The code below allows the tag to be formatted or something, and then, the above code wont receive an error.

///// ------ Function for formatting tag ------ \\\\\
export const formatTag = (test) => {
    test(true)
     // First request (erase tag)
     NfcManager.unregisterTagEvent()
     .then(() => NfcManager.registerTagEvent())
     .then(() => {
   NfcManager.requestNdefWrite(null, {format:true})
       .then(() => {
           alert('format completed')
           test(false)
        }).catch((err) => {
            alert('error: ZAP-D_101', err)
           test(false)
        })
    
    })
       .catch(err => {
           console.log('arrrrir', err);
           test(false)
         });
} 
///// -----------------       ----------------- \\\\\

Im not sure why you are receiving this error. Im sorry if this doesn't help, as I said, I am fairly new myself and haven't seemed to fully wrap my head around this code. Hopefully someone can jump in and share ther thoughts

junaidrazaq, 2020-12-19:
Also, this is where the function is being called:

{/* -------- FORMAT TAG -------- */}
      <Text style={{all my styles in here}}>
        If your having any problems, format the tag using the button below and try
        to configure your tag again 
      </Text>

      <TouchableHighlight
        style={{ all my styles in here }}
        onPress={() => formatTag(setIsWifiTestRunning)}>
        <Text style={styles.textStyle}>Format</Text>
      </TouchableHighlight>

{/* -------- LINK WITH WIFI -------- */}
      <Text style={{styles}}>
        If you'd like to configure Your tag link with your wifi, click here
      </Text>

      <TouchableHighlight
        style={{ my styling in here}}
        onPress={() => writeTag(1, wifiName, wifiPassword, setIsWifiTestRunning, setWifiName, setWifiPassword)}>
        <Text style={styles.textStyle}>WIFI</Text>
      </TouchableHighlight>

{/* -------- WRITE TO TAG -------- */}
      <Text style={{textAlign:'center', paddingTop:20, paddingBottom:15, width:'75%'}}>
        If Nfc is enabled, go ahead and click write and place your tag behind your device
      </Text>

      {!isTestRunning ?
        <TouchableHighlight
        style={{ styles here}}
        onPress={
          Platform.OS === 'android' ?
        () => writeTag(0, 'mywebsite.com/' + username, null, setIsTestRunning)
        : null
      }>
        <Text style={styles.textStyle}>WRITE</Text>
      </TouchableHighlight>
      : null
      }

I took out my styling to make it a lil cleaner

junaidrazaq, 2020-12-19:
So I also have code which is used when on ios, however, at the time of receiving this empty tag not writing error, I was editing on android. I assumed the issue was also with ios, but I just checked and it seems that this particular error does not occur on ios thankfully.

In case you want the code for ios, here it is:

///// --- Function for writing tag IOS --- \\\\\

  _cleanUp = () => {
    NfcManager.cancelTechnologyRequest().catch(() => 0);
  }
  _testNdef = async (textToWrite) => {
    try {
      let resp = await NfcManager.requestTechnology(NfcTech.Ndef, {
        alertMessage: 'Ready to write some NFC tags!'
      });
      console.warn(resp);
      let ndef = await NfcManager.getNdefMessage();
      console.warn(ndef);
      let bytes = buildUrlPayload(textToWrite);
      await NfcManager.writeNdefMessage(bytes);
      console.warn('successfully write ndef');
      await NfcManager.setAlertMessageIOS('Your tag is now linked!');
      this._cleanUp();
    } catch (ex) {
      console.warn('ex', ex);
      this._cleanUp();
    }
  }

I am yet to configure a wifi configuration on ios, which is why it looks a little different to my android code

avivku90, 2021-02-03:
i use the same code for writing from IOS
but it won't let me write on an empty tag (ios14+)
does it work for you?
and thank you for your reply

junaidrazaq, 2021-02-23:
So I recently updated the version to the latest, and my code no longer worked. Therefore, I downgraded back down to version 2.2.0.
This version allows the code to run successfully. Sorry for the late reply, I hope this helps

closed 2021-05-26 by github-actions[bot]