revtel/react-native-nfc-manager #350: On Android, how do I launch an app through NFC

yfs-2000, 2020-12-22:
I'm a newbie and I know it's not a plugin problem but it bothers me
I tried the android/app/AndroidManifest. XML directory to add this code, but no effect, I don't know where is the problem, hope someone can help me, thank you very much
` <activity android:name="ShareActivity"   android:launchMode="singleTask">
            <intent-filter>
                <action android:name="android.nfc.action.NDEF_DISCOVERED"/>
                <category android:name="android.intent.category.DEFAULT"/>
            </intent-filter>
        </activity>`

whitedogg13, 2021-02-20:
Hi @yfs-2000 

there're several approach for this, the simplest one is based on Android's AAR system:
* doc [here](https://developer.android.com/guide/topics/connectivity/nfc/nfc)
    * Please remember the data in AAR should be your app's `package name`
* Most well-known NFC tool applications (such as NXP Tag Writer / NFC Tools) support writing AAR into NFC tags, as well as our library.

Or, you can use custom scheme, it's slightly more complex. The steps are:

1. add following intent-filter into your AndroidManifest:
```xml
<intent-filter>
    <action android:name="android.intent.action.VIEW" />
    <category android:name="android.intent.category.DEFAULT" />
    <category android:name="android.intent.category.BROWSABLE" />
    <data android:scheme="com.myapp.pkgname" android:host="*" />
</intent-filter>
```
PS: Please replace `com.myapp.pkgname` into your app's package name.

2. For your NFC tag, you should write RTD_URI with this value `com.myapp.pkgname://whatever` (again, please replace `com.myapp.pkgname` into your package name.

I will close this for now, please feel free to re-open or fire another one if the issue remains.

closed 2021-02-20 by whitedogg13

mthines, 2021-06-27:
@whitedogg13 Firstly, thanks for this super awesome library 💪 

## I have two questions/requests.

### 1. Could you provide some more information on the write process for this?

At [`developer.android`](https://developer.android.com/guide/topics/connectivity/nfc/nfc) they write the following: "All you need to do is embed the `AAR` anywhere in your `NdefMessage`"

#### Should that be when encoding the message, 
```tsx
const bytes = Ndef.encodeMessage([Ndef.textRecord(message)]);
```

#### or writing the NdefMessage?
```tsx
await NfcManager.ndefHandler.writeNdefMessage(bytes);
```

#### Full context
```tsx
try {
  await NfcManager.requestTechnology(NfcTech.Ndef, {
    alertMessage: t('nfc:writeMessage', { type }),
  });

  const message = JSON.stringify({ type, value });
  const bytes = Ndef.encodeMessage([Ndef.textRecord(message)]); // <---

  if (bytes) {
    await NfcManager.ndefHandler.writeNdefMessage(bytes); // <---

    if (Platform.OS === 'ios') {
      await NfcManager.setAlertMessageIOS(t('common:success'));
    }

    response = true;

    setStatus('success');
  }
} catch (ex) {
  handleException(ex);
  setStatus('error');
} finally {
  handleCancel();
}
```

### 2. Regarding the package name.

Do it need to be suffixed with `.pkgname`, and does it need to match something somewhere?

It would be super nice to have an code example of this :) 

Thanks!