revtel/react-native-nfc-manager #352: ios no tech supported

FirassMraihi, 2020-12-29:
Greeting i'm trying to scan nfc tag with ios it always display no tech supported
i've used : 
IsoDep ,
MifareClassic ,
MifareUltralight,
MifareIOS ,
Iso15693IOS ,
FelicaIOS

SmartArray, 2020-12-31:
Please give us your complete code.

I have used some of those with my iPhone X (iOS 13) and it worked flawlessly.

alikianinejad, 2020-12-31:
same problem on IOS 14

FirassMraihi, 2021-01-02:
import React, {useState, useEffect} from 'react';
import {View, Image, ToastAndroid, Platform, Alert} from 'react-native';
import {connect} from 'react-redux';
import styles from './styles';
import Button from '../app/button';
import Header from '../../components/app/header';
import TitleText from '../../components/app/titleText';
import Container from '../../components/app/container';
import i18n from '../../translation';
import {getChoiceUser} from '../../redux/actions/withoutNFC';
import {saveNFCData} from '../../redux/actions/app';
import NfcManager, {NfcEvents, NfcTech} from 'react-native-nfc-manager';
import {
arrayBufferToHexString,
dataExtraction,
hexStringToArrayBuffer,
hex_to_ascii,
} from '../../utils/nfcHandler';
const {t} = i18n;
const ScanNFC = ({route, navigation, getChoiceUser, saveNFCData}) => {
const [nfcData, setNfcData] = useState(null);

const _comunicate = async () => {
let machineInfo = {};
let tech = Platform.OS === 'ios' ? NfcTech.MifareClassic : NfcTech.NfcV;
console.log('teeeeecch', tech);
NfcManager.setEventListener(NfcEvents.DiscoverTag, tag => {
console.log('tag***', tag);
NfcManager.unregisterTagEvent().catch(() => 0);
});
let resp = await NfcManager.requestTechnology(tech);
let tag = await NfcManager.getTag();
console.log('****$$$$',tag )
let tagId = tag.id;
machineInfo.tagId = tag.id;
try {
let message = '2223' + tagId + '0004';
let arrayBuffer = hexStringToArrayBuffer(message);
let byteArray = new Int8Array(arrayBuffer);
if (Platform.OS === 'ios') {
resp = await NfcManager.sendMifareCommandIOS(Array.from(byteArray));
} else {
resp = await NfcManager.transceive(Array.from(byteArray));
}
let result = arrayBufferToHexString(resp);
let machineData = hex_to_ascii(result.substring(10));
machineInfo = dataExtraction(machineData, tagId);
ToastAndroid.show(t('nfcScan.scanSucess'), ToastAndroid.LONG);
setNfcData(machineInfo);
} catch (ex) {
ToastAndroid.show(t('nfcScan.scanFailed'), ToastAndroid.LONG);
}
_cancel();
_cleanUp();
_startNfc();
_comunicate();
};
const _startNfc = () => {
NfcManager.start()
.then(() => {
NfcManager.isSupported(NfcTech.MifareClassic)
.then(() => console.log('Mifare classic is supported'))
.catch(err => console.log('******', err))
})

};
useEffect(() => {
_startNfc();
_comunicate();
return _cleanUp();
}, []);
const _cancel = () => {
NfcManager.unregisterTagEvent();
};
const _cleanUp = () => {
NfcManager.cancelTechnologyRequest().catch(() => 0);
};
return (
<Container>
<Header simpleHeader />
<TitleText text={t('interventions.scanNFCText')} />
<View style={styles.scanNFCImageContainer}>
<Image
source={require('../../assets/scanNfc.png')}
style={styles.scanNFCImage}
/>
</View>
<View style={styles.scanNFCButtons}>
<Button
label={t('app.cancel')}
style={styles.cancelButton}
labelStyle={styles.ignorLabelButton}
onPress={() => {
_cancel();
_cleanUp();
navigation.goBack();
}}
/>
<Button
label={t('interventions.enterManually')}
style={styles.enterManuallyButton}
labelStyle={styles.enterManuallyLabelButton}
onPress={() => {
if (nfcData) {
saveNFCData(nfcData);
navigation.navigate(
`${
route.params.sceneToNavigate
? route.params.sceneToNavigate
: 'accessCondition'
}`,
{
title: 'Cadrer et éviter la présence d’individus',
reappro: route.params.reappro,
inactive: route.params.inactive,
newIntervention: route ? route.params.newIntervention : false,
type: 'oui',
},
);
getChoiceUser();
} else {
Alert.alert(
"Le Tag NFC n'a pas été détecté.Veuillez résayer",
'',
[{text: 'OK', onPress: () => console.log('cancel')}],
{cancelable: false},
);
}
}}
/>
</View>
</Container>
);
};
export default connect(
null,
{
getChoiceUser, saveNFCData
},
)(ScanNFC);

closed 2021-05-26 by github-actions[bot]