revtel/react-native-nfc-manager #353: NFC is not supported on this Device (ios 14, IPHONE xs)

aminvb12, 2020-12-31:
I tried to run example code which is declared on your documentation [example](https://github.com/whitedogg13/react-native-nfc-manager/tree/master/example) but I've got `NFC is not supported`, although I've followed this link, I've coped with mentioned issue.

RocKer004, 2021-01-21:
I get the same result on every device, android and ios, even though NFC is supported by my Samsung Galaxy a51

whitedogg13, 2021-01-21:
hi guys

Please try this project to see if it works: https://github.com/revtel/react-native-nfc-rewriter

simply do:

1. clone the repo

2. npm install

3. cd ios && pod install

4. open ios/NfcOpenReader.xcworkspace, run the app into real device

It should run properly, at least for my case

If you're using slack, you can also join this workspace: https://join.slack.com/t/reactnativenf-ewh2625/shared_invite/zt-l2phclhi-AoYGQOOlCmvCSa3sBiA0IQ

and you can send me DM about the issue, it's easier to chat on slack.

RocKer004, 2021-01-21:
thanks a lot for replying, I'll try right now. Also, joined slack

YusufBesim, 2021-03-20:
Hello, still not working. tried react-native-nfc-rewrite repo to. Stuck on write alert

platform: IOS 14
device: iPhone 12 Pro

VentroArt, 2021-04-05:
Save issue 
device: IPhone Xr
platform: IOS 14.4.2

Receiving this on `NfcManager.start();` call

ishan5ain, 2021-06-23:
I'm experiencing the same issue. Even running the NFC ReWrite test app from App Store doesn't do anything when trying to write NDef messages. Tested on iOS version 14.6 and 12.1.4. 

Additionally, Not sure what is the root cause here is but browsing documentation from Apple on Core NFC sample app example code [here](https://developer.apple.com/documentation/corenfc/building_an_nfc_tag-reader_app). The documentation talks about: 
> To write to a tag, the sample app starts a new reader session. This session must be active to write an NDEF message to the tag

I'm wondering if something similar needs to happen when implementing the 'writeNDef' code in this library.

whitedogg13, 2021-06-25:
@ishan5ain Hi, can you tell me what tag do you use?

ishan5ain, 2021-06-25:
@whitedogg13 Hey, I'm using [PN532 NFC/RFID controller breakout board](https://www.adafruit.com/product/364) board from Adafruit

closed 2021-09-09 by github-actions[bot]