revtel/react-native-nfc-manager #356: Where should I get my tag data ?

Enzo-ltz, 2021-01-22:
Hello ! 

I'm actually working on NFC with react-native and I'm now getting the following problem :

I used NFC Writter example to write an empty tag. This work since I get the correct result when I scan it with NFC Tools.
Here's what I write on my tag :
[image]

However, with the NFC Reader example, I can't find the same data ! Here's what I can read : 
[image]

I would like to be able to find what I write ... Thank's for helping !

whitedogg13, 2021-01-24:
Hi @Enzo-ltz your data is inside the `ndefMessage's payload` field. 

You will need to decode it to see the content, the code is something like this:
```javascript
import {Ndef} from 'react-native-nfc-manager';
...
Ndef.uri.decodePayload(ndefMessage[0].payload);
...
```

closed 2021-02-10 by whitedogg13

TommysG, 2024-02-15:
> The above kind of works, but it should be in the documentation, clearly stated! Also, because we are using `uri`, it adds "https://www.en" in the front? How do we get just text?

You can use `Ndef.text.decodePayload` instead. I have this function in my code:

```ts
export const parseText = (tag: any) => {
  try {
    if (Ndef.isType(tag.ndefMessage[0], Ndef.TNF_WELL_KNOWN, Ndef.RTD_TEXT)) {
      return Ndef.text.decodePayload(tag.ndefMessage[0].payload);
    }
  } catch (e) {
    console.log(e);
  }
  return null;
};

```

and use it in my code like this:
```ts 
const tagData = parseText(tag) ?? tag?.id;
```

I hope that helps!!!