revtel/react-native-nfc-manager #357: Writing data to sectors and blocks / Writing Non-NDEF NFC Tags

RocKer004, 2021-01-24:
Hi there, I was requested to make an app to write to Non-NDEF NFC Tags. The device (card reader) can only read data from the cards sectors/blocks and I need to emulate that somehow with the app. NFCManager has the method mifareClassicWriteBlock, but it doesn't seem to be working so well. Any kind of help will be highly appreciated. Thanks in advance !

whitedogg13, 2021-02-10:
Hi @RocKer004 if you're targeting mifare classic, please note that it won't work on iOS. For Android, it should work properly since Android provides some native APIs to directly deal with it.

Since this topic is kinda verbose and strongly related to your use case, please consider join our slack channel for more detailed discussion: 
* Join here: https://join.slack.com/t/reactnativenf-ewh2625/shared_invite/zt-l2phclhi-AoYGQOOlCmvCSa3sBiA0IQ

I will close it for now, please feel free to reopen or fire another one if there're still issues.

closed 2021-02-10 by whitedogg13

melodyadenby, 2024-04-19:
If anyone is scrolling along, this is how I write to specific blocks for the NTAG 215. Please note each block is 4 in length!
`async function writeBlock(text: string, block: number) {
    try {
      // Prepare the data
      const data = text
        .split("")
        .map(char => (char === " " ? 0 : char.charCodeAt(0)));
      while (data.length < 4) {
        data.push(0); // Pad the data to fill 4 bytes as needed
      }

      // Ensure no previous technology requests are active
      await NfcManager.cancelTechnologyRequest();

      // Request the technology (NfcA is typical for Mifare Ultralight and many Type 2 tags)
      await NfcManager.requestTechnology(NfcTech.NfcA);

      // Command to write to block 4, assuming MIFARE Ultralight command format
      const command = [0xa2, block, ...data];
      const response = await NfcManager.transceive(command);
      console.log("Response:", response);

      // Check if the write was successful
      if (response[0] !== 0x0a) {
        throw new Error("Fail to write");
      }
    } catch (err) {
      console.warn(`Error on writing ${text} to block 4: ${err}`);
    } finally {
      // Ensure the NFC tech request is cancelled after the operation
      await NfcManager.cancelTechnologyRequest();
    }
  }
  async function write(text: string, startingBlock: number) {
    let chunks = text.match(/.{1,4}/g);
    // Check if chunks is null
    if (!chunks) {
      console.error("Text provided cannot be split into chunks, or is empty.");
      return;
    }

    // Write each chunk to consecutive blocks starting from startingBlock
    for (let i = 0; i < chunks.length; i++) {
      await writeBlock(chunks[i], startingBlock + i);
      setTimeout(() => {
        console.log(`Wrote ${chunks[i]}`);
      }, 100);
    }
  }`
  
  And this is how you call it: `write("hello world", 4) //converts spaces to zeros`