revtel/react-native-nfc-manager #359: Can we erase a tag or overwrite a tag using the library?

MaoJJJJJ, 2021-01-27:
Hi @whitedogg13,

Thank you for this library, it works really well! But during the writing process, seems like as long as the tag has existing payload, it will throw error says writing failed. So I wonder if there is a way to erase the tag for rewriting or there is a way for overwriting.

Cheers.

closed 2021-02-03 by MaoJJJJJ

klcantrell, 2022-01-08:
@MaoJJJJJ did you close this issue because you found a way to erase a tag with this library? If so, do you mind if I ask, how did you manage to do it?

whitedogg13, 2022-01-09:
@klcantrell it's strongly related to the NFC tag you're using. There's no platform API to simply erase every kind of NFC tag.

However, if you're targeting T2T tags such as NXP NTAG 21x family, take a look as [this code](https://github.com/revtel/react-native-nfc-rewriter/blob/master/src/NfcProxy.js), the `eraseNfcA` function will similar to what you need.

klcantrell, 2022-01-09:
Thank you for the explanation and very useful library, @whitedogg13!