revtel/react-native-nfc-manager #360: NfcManager.requestTechnology doesn't detect tags on IOS 14.4

ChristopheBelpaire, 2021-01-28:
Hello, 
I'm trying to read tag with the following code : 
        let resp = await NfcManager.requestTechnology([NfcTech.MifareIOS, NfcTech.Iso15693IOS, NfcTech.IsoDep, NfcTech.Ndef]);
        console.warn(resp);
        let tag = await NfcManager.getTag();
        console.warn(tag);
And they aren't detected.
If I use : 
 NfcManager.setEventListener(NfcEvents.DiscoverTag, tag => { console.log(tag)})
It works fine (my tags are ndef formated).

I need to use the request technology to get the UID of the tag.
I also tried your react-native-nfc-rewriter app and I could use the "Read ndef" (witch use the setEventListener method)
but the "Scan nfc tag" doesn't detect my tags.

All of my tags are readable with nxp tag info, and I can get the uids on this app.

Any idea ?

Thanks in advance!

whitedogg13, 2021-01-30:
Hi @ChristopheBelpaire thanks for reporting.

I just find an issue maybe related to this, can you remove `NfcTech.Ndef` from `requestTechnology` and try again? The issue might be fixed by this workaround, and we can still get NDEF via `getTag` call, like this:

```javascript
let resp = await NfcManager.requestTechnology([NfcTech.MifareIOS, NfcTech.Iso15693IOS, NfcTech.IsoDep]);
console.warn(resp);
let tag = await NfcManager.getTag();
console.warn(tag);
```

ChristopheBelpaire, 2021-02-01:
Hello @whitedogg13 thanks for the answer.
I tried your suggestion without success.
Here is the tag info of my tag :

````
** TagInfo scan (version 4.24.7) 2021-02-01 09:40:54 **
Report Type: External

-- IC INFO ------------------------------

# IC manufacturer:
EM Microelectronic-Marin SA

# IC type:
Unknown IC implementing Iso 15693

# NFC Forum NDEF-enabled tag:
Unknown Type Tag

-- NDEF ------------------------------

# NFC data set information:
NDEF message containing 1 record
Current message size: 105 bytes
Maximum message size: 146 bytes
NFC data set access: Read & Write

# Record #1: URI record:
Type Name Format: NFC Forum well-known type
Short Record
type: "U"
protocol field: https://
URI field: cloud.genexir.selinko.com/j0U7Gcgr-zH1Z45G?uid=4BZ4ARAaXWU&token=AAJl&f=g&sig=fo9rG-ZDVmv9_jDUYODY9Q
Payload length: 101 bytes
Payload data:

[00] 04 63 6C 6F 75 64 2E 67 65 6E 65 78 69 72 2E 73 |.cloud.genexir.s|

[10] 65 6C 69 6E 6B 6F 2E 63 6F 6D 2F 6A 30 55 37 47 |elinko.com/j0U7G|

[20] 63 67 72 2D 7A 48 31 5A 34 35 47 3F 75 69 64 3D |cgr-zH1Z45G?uid=|

[30] 34 42 5A 34 41 52 41 61 58 57 55 26 74 6F 6B 65 |4BZ4ARAaXWU&toke|

[40] 6E 3D 41 41 4A 6C 26 66 3D 67 26 73 69 67 3D 66 |n=AAJl&f=g&sig=f|

[50] 6F 39 72 47 2D 5A 44 56 6D 76 39 5F 6A 44 55 59 |o9rG-ZDVmv9_jDUY|

[60] 4F 44 59 39 51                                  |ODY9Q           |

# NDEF message:
[00] D1 01 65 55 04 63 6C 6F 75 64 2E 67 65 6E 65 78 |..eU.cloud.genex|

[10] 69 72 2E 73 65 6C 69 6E 6B 6F 2E 63 6F 6D 2F 6A |ir.selinko.com/j|

[20] 30 55 37 47 63 67 72 2D 7A 48 31 5A 34 35 47 3F |0U7Gcgr-zH1Z45G?|

[30] 75 69 64 3D 34 42 5A 34 41 52 41 61 58 57 55 26 |uid=4BZ4ARAaXWU&|

[40] 74 6F 6B 65 6E 3D 41 41 4A 6C 26 66 3D 67 26 73 |token=AAJl&f=g&s|

[50] 69 67 3D 66 6F 39 72 47 2D 5A 44 56 6D 76 39 5F |ig=fo9rG-ZDVmv9_|

[60] 6A 44 55 59 4F 44 59 39 51                      |jDUYODY9Q       |

# NDEF Capability Container (CC):
Mapping version: 1.0
Maximum NDEF data size: 152 bytes
NDEF access: Read & Write
Supports Multiple Block Read
[0] E1 40 13 01                                     |.@..            |

-- EXTRA ------------------------------

# Memory size:
152 bytes
* 38 blocks, with 4 bytes per block

# IC detailed information:
Supported read commands:
* Single Block Read
* Multiple Block Read
* Get Multiple Block Security Status
* Get System Information
AFI supported
DSFID supported
IC reference value: 0x1E
Customer ID: 0x001

-- FULL SCAN ------------------------------

# Technologies supported:
ISO/IEC 15693-3 compatible
ISO/IEC 15693-2 compatible

# Android technology information:
Tag description:
* TAG: Tech [android.nfc.tech.NfcV, android.nfc.tech.Ndef]
* Maximum transceive length: 253 bytes

# Detailed protocol information:
ID: E0:16:78:01:10:1A:5D:65
AFI: 0x00
DSFID: 0x00

# Memory content:
[00] . E1 40 13 01 |.@..|
[01] . 03 69 D1 01 |.i..|
[02] . 65 55 04 63 |eU.c|
[03] . 6C 6F 75 64 |loud|
[04] . 2E 67 65 6E |.gen|
[05] . 65 78 69 72 |exir|
[06] . 2E 73 65 6C |.sel|
[07] . 69 6E 6B 6F |inko|
[08] . 2E 63 6F 6D |.com|
[09] . 2F 6A 30 55 |/j0U|
[0A] . 37 47 63 67 |7Gcg|
[0B] . 72 2D 7A 48 |r-zH|
[0C] . 31 5A 34 35 |1Z45|
[0D] . 47 3F 75 69 |G?ui|
[0E] . 64 3D 34 42 |d=4B|
[0F] . 5A 34 41 52 |Z4AR|
[10] . 41 61 58 57 |AaXW|
[11] . 55 26 74 6F |U&to|
[12] . 6B 65 6E 3D |ken=|
[13] . 41 41 4A 6C |AAJl|
[14] . 26 66 3D 67 |&f=g|
[15] . 26 73 69 67 |&sig|
[16] . 3D 66 6F 39 |=fo9|
[17] . 72 47 2D 5A |rG-Z|
[18] . 44 56 6D 76 |DVmv|
[19] . 39 5F 6A 44 |9_jD|
[1A] . 55 59 4F 44 |UYOD|
[1B] . 59 39 51 FE |Y9Q.|
[1C] . 00 00 00 00 |....|
[1D] . 00 00 00 00 |....|
[1E] . 00 00 00 00 |....|
[1F] . 00 00 00 00 |....|
[20] . 00 00 00 00 |....|
[21] . 00 00 00 00 |....|
[22] . 00 00 00 00 |....|
[23] . 00 00 00 00 |....|
[24] . 00 00 00 00 |....|
[25] . 00 00 00 00 |....|

  x:locked, .:unlocked

--------------------------------------
```
Any idea here I can debug the library in order to see where the tag not detected ?

Thanks in advance!

MatteoAlberghini, 2021-02-12:
Hi, I have the same problem. 
```
let resp = await NFCManager.requestTechnology([
  NfcTech.MifareIOS,
  NfcTech.Iso15693IOS,
  NfcTech.IsoDep,
  NfcTech.Ndef,
]);
console.log(resp);
return await NFCManager.getTag();
```

The tags are found on android, and I can send IsoDep commands to them (such as making auth and changing auth keys) using transceive.
I'm using ntag 424 DNA: https://www.nxp.com/docs/en/data-sheet/NT4H2421Gx.pdf

MatteoAlberghini, 2021-02-16:
I want to add to this issue: I solved the problem. In my instance, the problem was that I didn't specify NFC ISO 7816 application identifiers inside the .plist, as per this docs https://developer.apple.com/documentation/corenfc/nfciso7816tag . 
I found another problem though: If I run Request Techonology with IsoDep and Ndef I get a "React Native callback type only permits a single invocation from native code". I need to find only IsoDep tags so is no big deal, but might still be a problem for others.

closed 2021-05-26 by github-actions[bot]