revtel/react-native-nfc-manager #361: MifareUltralight and sendMifareCommandIOS

zabojad, 2021-02-01:
Hi there!

Does `sendMifareCommandIOS` would work with a `MifareUltralight` tag?

My tag  supported techs : `"techTypes": ["android.nfc.tech.NfcA", "android.nfc.tech.MifareUltralight", "android.nfc.tech.Ndef"],`

I need to send two commands to that tag that I can send successfully with `transceive` on Android.

I was wondering if I could just use `sendMifareCommandIOS` the same way I use `transceive` on Android. Is that possible?

If yes, should I "requestTechnology" `NfcTech.MifareIOS` of `NfcTech.MifareUltralight`?

Thank you for your attention.

whitedogg13, 2021-02-01:
Hi @zabojad you are right, `sendMifareCommandIOS` is just like `transceive` for MifareUltralight tags. 

the code probably looks like this:
```javascript
// requesting technology
if (Platform.OS === 'ios') {
  await NfcManager.requestTechnology([NfcTech.MifareIOS]);
} else {
  await NfcManager.requestTechnology([NfcTech.Ndef]);
}     

// send commands
if (Platform.OS === 'ios') {
  resp = await NfcManager.sendMifareCommandIOS(payload);
} else {
  resp = await NfcManager.transceive(payloads);
}   
```

BTW, you can also check some example from this repo: http://github.com/revtel/react-native-nfc-rewriter

Thanks!

whitedogg13, 2021-02-03:
Close for now, feel free to reopen or fire another one if you still have problems!

closed 2021-02-03 by whitedogg13

zabojad, 2021-02-03:
hi @whitedogg13 !
Sorry for not answering earlier.
Thank a lot for your reply and the great job you provide on this module!

zabojad, 2021-02-03:
Hi @whitedogg13 !

Sorry to come back to this again but  when I try to use `NfcTech.MifareIOS` with my tag (that supports MifareUltralight), I  get : `"No tech matches"`:

When looking at the ios code below, it seems to be that `[tagType isEqualToString:requestType]` will never  be true:
```
- (void)tagReaderSession:(NFCTagReaderSession *)session didDetectTags:(NSArray<__kindof id<NFCTag>> *)tags
{
    NSLog(@"NFCTag didDetectTags");
    if (@available(iOS 13.0, *)) {
        if (techRequestCallback != nil) {
            BOOL found = false;
            RCTResponseSenderBlock pendingCallback = techRequestCallback;
            
            // by setting callback to nil, we know the promise is resolved
            techRequestCallback = nil;

            for (NSString* requestType in techRequestTypes) {
                for (id<NFCTag> tag in tags) {
                    NSString * tagType = [self getRNTechName:tag];
                    // here we treat Ndef is a special case, because all specific tech types
                    // inherites from NFCNDEFTag, so we simply allow it to connect
                    if ([tagType isEqualToString:requestType] || [requestType isEqualToString:@"Ndef"]) {
                        [sessionEx connectToTag:tag
                              completionHandler:^(NSError *error) {
                            if (error != nil) {
                                pendingCallback(@[getErrorMessage(error)]);
                                return;
                            }
                            
                            pendingCallback(@[[NSNull null], requestType]);
                        }];
                        found = true;
                        break;
                    }
                }
            }
            
            if (!found) {
                pendingCallback(@[@"No tech matches", [NSNull null]]);
            }
        }
    }
}
```

Is that a bug? `MifareIOS` should work with `MifareUltralight` is that right?

whitedogg13, 2021-02-03:
@zabojad the native code should be fine, the `MifareIOS` in react-native-nfc-manager actually represent for iOS's [NFCMifareTag](https://developer.apple.com/documentation/corenfc/nfcmifaretag?language=objc), and it should cover MifareUltralight, as listed [here](https://developer.apple.com/documentation/corenfc/nfcmifarefamily?language=objc).

However, I haven't tested on any MifareUltralight tags yet.

I think maybe you can set a breakpoint in XCode, to see what `tag` type it actually is, in the code you listed:
```objective-c
- (void)tagReaderSession:(NFCTagReaderSession *)session didDetectTags:(NSArray<__kindof id<NFCTag>> *)tags
{ ... }
``` 
the `tags` sent from iOS is an array of [NFCTag](https://developer.apple.com/documentation/corenfc/nfctag?language=objc) object, which contains a `type` property and the possible `type` value is listed in Apple's [doc](https://developer.apple.com/documentation/corenfc/nfctagtype?language=objc).

BTW, we have slace channel about this library, you're welcome to join and we can discuss more efficiently there: https://join.slack.com/t/reactnativenf-ewh2625/shared_invite/zt-l2phclhi-AoYGQOOlCmvCSa3sBiA0IQ

zabojad, 2021-02-03:
Excellent, thank you @whitedogg13 for your answer.

I'm waiting for a new iPhone enabled NFC from my company. I'll keep you updated on this issue as soon as I have received it and tested/debug it...