revtel/react-native-nfc-manager #362: Delegate does not implement -readerSessionDidBecomeActive: method

kaganerbay, 2021-02-02:
Hi,
I tried the v2-ios+android-read-ndef example on IOS 14 real device but cannot read any NFC. Xcode gives me error: **[CoreNFC] 00000002 828465b0 -[NFCNDEFReaderSession _callbackDidBecomeActive]:218  Delegate does not implement -readerSessionDidBecomeActive: method .**

I think, I cannot get in the setEvetListener method and so cannot get the tag.

Does anyone has a solution for this?

Thanks in advance!

whitedogg13, 2021-02-03:
@kerbay Thanks for reporting this. Can you report the Xcode version and the iOS version / device you're using?

kerbay, 2021-02-03:
Of course,
- Xcode version is 12.4
- Device is iPhone 7 and iOS version 14.3
- React Native version 0.63.3

Thanks.

whitedogg13, 2021-02-03:
Hi @kerbay Just tested. I can confirm this message did print out by Xcode, but this is simply a warning message rather than an error. This shouldn't be the reason why the NFC tag isn't detected. 

I will close this for now, since this shouldn't be the root cause. To troubleshoot this kind of issue, you can join our slack channel and we can chat there: https://join.slack.com/t/reactnativenf-ewh2625/shared_invite/zt-l2phclhi-AoYGQOOlCmvCSa3sBiA0IQ

closed 2021-02-03 by whitedogg13