revtel/react-native-nfc-manager #364: Android NfcV transceive doesn't work, iOS works

lukebars, 2021-02-03:
Hello,

I'm having issues reading NFCV(ISO15693) tag with android. iOS transceive works fine, but android always fails - ` ex transceive fail - Tag was lost.)
Android:
```
      const tech =  NfcTech.NfcV;

      const resp = await NfcManager.requestTechnology(tech, {
          alertMessage: 'Ready to do some custom NfcV cmd!'
      });

      const resp1 = await NfcManager.transceive([0x20,0x20,0,0,0,0,0,0,0,0,0]);

      console.warn(resp1);
```

iOS:

```
    const tech = NfcTech.Iso15693IOS;

    await NfcManager.requestTechnology(tech, {
      alertMessage: 'Ready to scan the tag',
    });
    
    const blockBytes: Array<number> = await handler.readSingleBlock({
      flags: Nfc15693RequestFlagIOS.HighDataRate,
      blockNumber,
    });
```

There is no documentation on android part how the NFCV technology api should be used. What am I doing wrong?

lukebars, 2021-02-03:
Turns out the proper way to read NfcV on android is:

```js
    await NfcManager.requestTechnology(NfcTech.NfcV, {
      alertMessage: 'Ready to write some NFC tags!'
    });

    let tag = await NfcManager.getTag();
    const idBytes = tag.id
      .split(/(..)/g)
      .filter(s => s);
    idBytes.forEach((val, i, a) => a[i] = parseInt(val, 16));

    let res = await NfcManager.transceive([32,32,...idBytes,0]); // 32 - read and high speed flags, then follows tag id and last byte is the byte address you want to read

```

I think this will be quite useful as an example.

closed 2021-02-03 by lukebars

whitedogg13, 2021-02-03:
@lukebars it's awesome that you figure it out! Since its definitely helpful to others, would you mind put it into example and make a PR?

lukebars, 2021-02-04:
@whitedogg13 sure, I'll raise a PR in a day or two