revtel/react-native-nfc-manager #366: How to read identity card&passport data

tcagkansokmen, 2021-02-05:
I'm using AppV2Apdu.js

but it returns complicated response. I need to write name-surname- photo etc.
How can I decrypt that?

tcagkansokmen, 2021-02-06:
I'm gonna loose my mind. really im gonna loose my mind.

my tag response is below:
```
applicationData: [] (0)

historicalBytes: [blabla, blabla, blabla, blabla, blabla, 101, 176, 133, 3, 1, …] (15)

id: "blabla"

initialSelectedAID: "A0000002471001"

tech: "IsoDep"
```

my NfcManager.sendCommandAPDUIOS([0xFF, 0xCA, 0x00, 0x00, 0x00]); response is blow:

`{response: [], sw1: 104, sw2: 129}`

whats wrong with it?
why it doesnt response me as "name: surname: cardid" etc?
how can I read all informations inside that passport nfc?

kerbay, 2021-02-08:
Hello,
I am working on a project to read eID card with NFC. I think firstly you need to do BAC(Basic access control). With that you need to provide document number, date of birth and expiry date to the card and then get the informations you want. However, I don't know how to do that with this package. If there is a way I am too happy to know.

closed 2021-05-26 by github-actions[bot]

LuuNguyen0811, 2023-03-03:
Do you have any solutions?

eobruno, 2024-07-02:
const apduCommand = [0x00, 0xA2, 0x00, 0x00, 0x10, ...bacKey];

response = await NfcManager.sendCommandAPDUIOS(apduCommand);

birkanozdamar, 2024-09-10:
> const apduCommand = [0x00, 0xA2, 0x00, 0x00, 0x10, ...bacKey];
> 
> response = await NfcManager.sendCommandAPDUIOS(apduCommand);
hi @eobruno 
How was it created bacKey ?

 const bacAuthenticationCommand = [
                0x00, 0xA2, 0x00, 0x00, 0x10,
                ...bacKey
            ];  
response = await NfcManager.transceive(bacAuthenticationCommand);

For BACKey, there is no more detail with the following method;

const createBACKey = async ({ serialNumber, birthDate, expiryDate }) => {
        const keySeed = serialNumber + birthDate + expiryDate;
        const digest = await Crypto.digestStringAsync(Crypto.CryptoDigestAlgorithm.SHA1, keySeed);
        const K_enc = digest.slice(0, 16);
        const K_mac = digest.slice(16, 32);

        return Buffer.from(K_enc + K_mac, 'hex'); 
    };