revtel/react-native-nfc-manager #368: Unable to write immeadiately after a read without having to physically move the device away (v1 partially works, not v2)

PedroMD, 2021-02-09:
Hello there @whitedogg13 

First off, thanks for this lib! It's been very helpful so far.

We've been using v2 in a new project and we had a use case which mandated an immediate READ after a WRITE. Even though we managed to solve it using v1.1.3 as per https://github.com/whitedogg13/react-native-nfc-manager/issues/120 (exactly the same use case as the one listed by the author), we now need to go further :) 

There are some flows in our app where we actually need to WRITE>READ>WRITE, which we've not been able to implement so far. Can you please point us in the right direction? Is there any particular sequence of funcs we should be calling after the 1st read in order to be able to trigger a successful write while not physically moving the device from the NFC tag? It does work, though, when we move the device away from the tag.

Is this something we should be able to achieve with v2? Maybe the v3 you're working on?

Any help would be much appreciated!
Thanks!

whitedogg13, 2021-02-09:
Hi @PedroMD sure, it’s totally possible through v2. 

I have made an example for you, please check: https://github.com/revtel/react-native-nfc-rewriter/tree/write-ndef-twice

Please check `writeNdefTwice` function in `src/NfcProxy.js`.
The code snippet for your question is listed here:
```javascript
  writeNdefTwice = async () => {
    let result = false;
    try {
      await NfcManager.requestTechnology(NfcTech.Ndef);

      let bytes;

      bytes = Ndef.encodeMessage([Ndef.textRecord('hello')]);
      await NfcManager.writeNdefMessage(bytes);

      const tag = await NfcManager.getTag();
      console.warn('tag', tag);

      bytes = Ndef.encodeMessage([Ndef.textRecord('hello again')]);
      await NfcManager.writeNdefMessage(bytes);

      result = true;
    } catch (ex) {
      console.warn(ex);
    }

    this.abort();
    return result;
  };
```

I will close this for now, please re-open or fire another one if there’re still issues. Thanks!

closed 2021-02-09 by whitedogg13

fgrs, 2021-06-16:
@whitedogg13 it's a good example that handles WRITE,WRITE scenario but what about READ,WRITE without the need of moving phone away?

whitedogg13, 2021-06-17:
@fgrs Sure. The response for `getTag()` will include the `ndefMessage`. Or, you can also read them explicitly through `getNdefMessage()`.

BTW, the following example is in v3.

```javascript
  firstReadThenWrite = async () => {
    try {
      await NfcManager.requestTechnology(NfcTech.Ndef);

      const tag = await NfcManager.getTag();
      console.warn('tag', tag); // tag will contain ndefMessage

      // in v3, we use .ndefHandler to access ndef related functions
      const resp = await NfcManager.ndefHandler.getNdefMessage();
      console.warn('ndef resp', resp);

      const bytes = Ndef.encodeMessage([Ndef.textRecord('hello')]);
      await NfcManager.ndefHandler.writeNdefMessage(bytes);
    } catch (ex) {
      console.warn(ex);
    } finally {
      NfcManager.cancelTechnologyRequest();
    }
  };

```

fgrs, 2021-06-25:
@whitedogg13 great example and it works but I don't think you can write after scan using background scan via:

`NfcManager.setEventListener(NfcEvents.DiscoverTag, async (tag) => {
[...]
})`

Can you correct me if I'm wrong? I tried all sort of different approaches but it always get's stuck on **NfcManager.requestTechnology(NfcTech.Ndef)** and I gave to tap twice on the Tag for the code to be fully executed.

fgrs, 2021-07-20:
I'm getting a little bit confused which way is best for detecting tags

    NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag) => {
      NfcManager.setAlertMessageIOS('NDEF tag found');
      NfcManager.unregisterTagEvent().catch(() => 0);
    });

or

    try {
      const resp = await NfcManager.requestTechnology(NfcTech.Ndef, {
        alertMessage: 'Ready to scan',
      });
      tag = await NfcManager.getTag();
    }

both seem to work similar with a little bit more impact on performance with #2 but writing after reading doesn't work with #1

    NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag) => {
      NfcManager.setAlertMessageIOS('NDEF tag found');
      const bytes = Ndef.encodeMessage([Ndef.textRecord('hello')]);
      await NfcManager.ndefHandler.writeNdefMessage(bytes);
      NfcManager.unregisterTagEvent().catch(() => 0);
    });

I think I got lost here, can you share your thought @whitedogg13

whitedogg13, 2021-07-20:
@fgrs Definitely use the second approach. (the `requestTechnology` one). The first one is only for simplest NDEF detecting, any app which requires more NFC feature than that should use `requestTechnology`. Thanks!

fgrs, 2021-07-20:
Thank you very much for the prompt reply!