revtel/react-native-nfc-manager #381: Android, Native Popup opens automatically saying new tag scanned when tag is scanned. How Can we disable this?

jungRoit, 2021-03-24:
Sometimes when scanning the NFC card a native popup saying New Tag Scanned appears and the functionality is not executed.

whitedogg13, 2021-03-25:
Android will fire `NDEF_DISCOVERED` intent once a NFC tag contains valid URI, or MIME NDEF record, if no applications register for the intent, the system one will popup. 

To fix this, you can call `NfcManager.start`, this API will register for this intent in foreground, so the system one won't popup.

whitedogg13, 2021-03-25:
I will close this for now, feel free to re-open or fire another one if there're still issues

closed 2021-03-25 by whitedogg13

jungRoit, 2021-03-25:
Thank you so much.

andreshsingh, 2022-10-30:
is it possible to disable the default scan ( "New tag scanned" ) before we call .start() ?

Sewb21, 2023-03-06:
@whitedogg13 I am calling start in my application to register the intent to the foreground app but I am still getting the "New Tag Collected" system popup and when the NFC tag contains a URL it's navigating to the browser. Is there a way of stopping this?

wordisthebird, 2023-09-21:
> @whitedogg13 I am calling start in my application to register the intent to the foreground app but I am still getting the "New Tag Collected" system popup and when the NFC tag contains a URL it's navigating to the browser. Is there a way of stopping this?

Any luck with this?

Sewb21, 2023-09-21:
@wordisthebird I did resolve this a while ago, IIRC it was the way the link was being written to the tag but I can't fully remember and I don't have access to the code base to check