revtel/react-native-nfc-manager #384: requestTechnology response error

kerbaypapilon, 2021-03-29:
Hi,
I am using 2.2.1 version of this module on RN 0.61.4. After I request a IsoDep technology, I got no response. However, I used this package with same version 1 month ago with RN 63.4, I was able to got response from tech and I read some tags. Any idea what is the problem?

`let tech = NfcTech.IsoDep;
      let resp = await NfcManager.requestTechnology(tech, {
        alertMessage: 'Please read the ID card',
      });`

I cannot get response from this..

whitedogg13, 2021-03-29:
Sorry, I have no idea, and I think the RN version shouldn't result in any differences like this.

kerbaypapilon, 2021-03-30:
It is totally my bad, I forgot to write;
`<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
	<array>
		<string>A0000002471001</string>
	</array>`
to info.plist file for reading ePassports. 

I am closing issue..

closed 2021-03-30 by kerbaypapilon