revtel/react-native-nfc-manager #385: Handling IOS Cancel Button

Damian333, 2021-04-01:
How can I handle "Cancel" button on IOS "listening for tag" pop up. 
Can I remove this pop up completely ?

Generally I have issue (different on Android and different on IOS) when I start listening for tags and stop it by "Cancel" button in ios or I use system back on Android. How can I check if request for technology is currently active and ask for technology only when it its not?

whitedogg13, 2021-04-08:
Hi @Damian333 

You won't be able to remove the iOS popup, however, you can detect the cancellation from users.

With `v3.2.0`, we have `NfcErrorIOS` to check for this:
```js
import NfcManager, {NfcTech, NfcErrorIOS} from 'react-native-nfc-manager';
...
    try {
      await NfcManager.requestTechnology([NfcTech.Ndef]);
      const tag = await NfcManager.getTag();
      console.warn(tag);
      await NfcManager.cancelTechnologyRequest();
    } catch (ex) {
      if (NfcErrorIOS.parse(ex) === NfcErrorIOS.errCodes.userCancel) {
        console.warn('user cancel NFC scanning');
      } else {
        console.warn(ex);
      }
    }
...
```

If you're in previous version, you can manually parse error string for "NFCError: 200" (200 is defined by iOS as user cancellation error code).

closed 2021-04-08 by whitedogg13

pke, 2024-06-13:
@whitedogg13 this does not work in the current version of the lib. When the user cancels or the session times out, the catch handler is not called.