revtel/react-native-nfc-manager #389: SessionClosed is not accurate and results in Duplicate registration error

sajadghawami, 2021-04-30:
Hey,

even though the `SessionClosed` event is fired, you will still get the following error:

```
Possible Unhandled Promise Rejection (id: 0):
"Duplicated registration"
```

i am using  `"react-native-nfc-manager": "^3.3.0"`.

Here is a small reproducable example:

```ts
export const useNDEF = () => {
  const [initializedNFC, setInitializedNFC] = React.useState(false);
  const cleanUp = () => {
    console.log("cleanedup");
    NfcManager.setEventListener(NfcEvents.DiscoverTag, null);
    NfcManager.setEventListener(NfcEvents.SessionClosed, null);
  };
  const readNFC = async () => {
    NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag: any) => {
      console.log(tag);
      NfcManager.setAlertMessageIOS("NDEF tag found");
      NfcManager.unregisterTagEvent().catch(() => 0);
      //   const decode = Ndef.uri.decodePayload(tag.ndefMessage[0].payload);
    });
    NfcManager.setEventListener(NfcEvents.SessionClosed, () => {
      cleanUp();
    });
    await NfcManager.registerTagEvent();
  };
  const initNFC = async () => {
    try {
      await NfcManager.start();
    } catch (e) {
      console.log("ERROR", e);
    } finally {
      setInitializedNFC(true);
    }
  };
  return { initNFC, readNFC, initializedNFC };
};
```

You need to call initNFC first.
After initialization call readNFC and when you see ` cleanedup` in the console, call readNFC again.

Any suggenstions or workarounds? I am using a timeout right now, which is not the most elegant solution.

whitedogg13, 2021-05-15:
@sajadghawami sorry for the late response.

I think to re-start NFC scanning without duplicated registration, we will have to make sure the previous `NfcEvents.SessionClosed` event is fired for the previous one.

If you'd like to use React Hook for this, I think a backing `ref` can be used. 

Here I have modified your original code a little bit to demonstrate this idea:

```javascript
export const useNDEF = () => {
  const hasActiveSessionRef = React.useRef(false); // <-- [CHECK HERE] to track whether we have active NFC session
  const [initializedNFC, setInitializedNFC] = React.useState(false);
  const cleanUp = () => {
    console.log("cleanedup");
    NfcManager.setEventListener(NfcEvents.DiscoverTag, null);
    NfcManager.setEventListener(NfcEvents.SessionClosed, null);
  };
  const readNFC = async () => {
    if (hasActiveSessionRef.current) { // <-- [CHECK HERE] check whether there's any existing session
      return;
    }

    NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag: any) => {
      console.log(tag);
      NfcManager.setAlertMessageIOS("NDEF tag found");
      NfcManager.unregisterTagEvent().catch(() => 0);
      //   const decode = Ndef.uri.decodePayload(tag.ndefMessage[0].payload);
    });
    NfcManager.setEventListener(NfcEvents.SessionClosed, () => {
      hasActiveSessionRef.current = false; // <-- [CHECK HERE] set active NFC session to false
      cleanUp();
    });
    await NfcManager.registerTagEvent();
    hasActiveSessionRef.current = true; // <-- [CHECK HERE] set active NFC session to true
  };
  const initNFC = async () => {
    try {
      await NfcManager.start();
    } catch (e) {
      console.log("ERROR", e);
    } finally {
      setInitializedNFC(true);
    }
  };
  return { initNFC, readNFC, initializedNFC };
};
```

I will close this issue for now, feel free to re-open or fire another one if there're still issues. Thanks~

closed 2021-05-15 by whitedogg13

sajadghawami, 2021-05-15:
@whitedogg13 

thank you very much for the response!

whitedogg13, 2021-05-16:
My pleasure~