revtel/react-native-nfc-manager #395: Incompatible documentation about eventListener

jaroslawkrol, 2021-05-13:
Hi,

There is a pretty fine solution how to detect NFC state changes, in your library. But current documentation in Readme is totally incorrect (probably based on older version ??), what cause devs (like me) have to find the correct way of use in code on their own, what is very annoying. 

Please, fix it.
Have a nice development this awesome lib! :)

whitedogg13, 2021-05-15:
@jaroslawkrol ya, thanks for bringing this. You're right and the documentation is indeed NOT clear enough. 

I'm still thinking about how to re-structure the documentation for this library. Since the NFC is such a huge topic to understand and lots of corner cases to consider, maybe re-write the whole documentation is required for the long run. However, as a solo developer (almost) with a day job, it's quite hard to keep track of all these.

I will keep this open and rename the title to "better documentation". I think we need a systematic improvement rather than any ad-hoc doc update.

Are you interested in helping us to improve that? :)

chazmcgrill, 2021-06-29:
Think jsdoc comments would be really useful on the exposed methods. Especially in files like `NfcManager.js`.

limal, 2021-07-23:
There used to be a list of all exposed functions in the README. I can't find even the most simplistic method like `isEnabled()` anywhere in the READMEs.

whitedogg13, 2023-03-07:
Hi @javix64 , it's awesome to hear that! I'm not active in twitter, so maybe we can discuss via email: whitedogg13@gmail.com. Thanks!