revtel/react-native-nfc-manager #398: Reading emulating tag

damianRayden, 2021-05-26:
I need to read tag emulated out of another android phone. Emulating is working fine, and I can read emulated tag with another phone with app TAG INFO NXP and everything is working ok. The problem is when I use for reading my own app which I have written with this library. When I am sticking phones together app instead of reading emulated tag turns on peer-2-peer transmission or launch preparing for payment system application. Is there a possibility to avoid this issue using this library or react-native? Or if I need to create native bridge? I think that generally this problem can be fixed if NXP TAG INFO can work.

whitedogg13, 2021-05-28:
@damianRayden Hi, please ty to call `NfcManager.setNdefPushMessage` and pass `null` into it, like this:

```javascript
NfcManager.setNdefPushMessage(null);
```

BTW, most NFC P2P features have been deprecated by Google, so we should rely on the above API but only use it to disable default Android Beam behavior.

damianRayden, 2021-05-28:
Thank you very much for response. 
Your solution works fine for blocking system reaction for peer-2-peer, but now the problem is that my app (written with react-native-nfc-manager) is reading emulated tag only when phone with emulated tag is has blocked screen (screen safer which) otherwise emulated tag is not read (on app TAG INFO NPX this example works fine). Instead of reading tag phone with tag emulation is doing some BEEP sound and that's all. Besides that I noticed issued that when I have more than one app which have tag emulation on the phone then during tag reading on phone with emulated tag appears popup with question about which app I want to use for process.

So now I have two questions. Can I get state on the phone with emulated tag which exist during screen safer mode? (or is there another solution for first problem mention above)
The second question is if my app can be the default one when another phone is reading the emulated tag?

closed 2021-08-13 by github-actions[bot]

cejuo, 2022-05-22:
> I need to read tag emulated out of another android phone. Emulating is working fine, and I can read emulated tag with another phone with app TAG INFO NXP and everything is working ok. The problem is [...]

Hey I'm looking for a way to emulate a tag with a phone. May I ask how are you emulanting it?