revtel/react-native-nfc-manager #404: [iOS] Not able to read MIFARE DESFire id

mrados7, 2021-06-07:
So I have tag when scanned by Android NFC Tools app says its type of `NXP MIFARE DESFire`. 
Technologies available are:
  *  IsoDep
  * NfcA
  * Ndef

 Android part works, tag is read and I get response like: 
```
{"canMakeReadOnly": false, "id": "0429771AC16980", "isWritable": true, "maxSize": 2046, "ndefMessage": [{"id": "", "payload": [Array], "tnf": 1, "type": [Array]}], "techTypes": ["android.nfc.tech.IsoDep", "android.nfc.tech.NfcA", "android.nfc.tech.Ndef"], "type": "NFC Forum Type 4"}
```

But iOS returns just: 
```
{"ndefMessage": [{"id": [], "payload": [], "tnf": 1, "type": []}]}
```
So I'm missing id and that's the only thing I need :D. Are there any workarounds for iOS, or some other React Native libs that may work? I know this is not iOS restriction cause other apps are able to get ID of tag.

Thanks in advance!

mrados7, 2021-06-07:
Nvm managed to do so with this code: 
```
 try {
      let tech = Platform.OS === "ios" ? NfcTech.MifareIOS : NfcTech.NfcA;
      let resp: any = await NfcManager.requestTechnology(tech, {
        alertMessage: "Ready to do some custom Mifare cmd!"
      });
      console.warn(resp);

      // the NFC uid can be found in tag.id
      let tag = await NfcManager.getTag();
      console.warn(tag);

      if (Platform.OS === "ios") {
        resp = await NfcManager.sendMifareCommandIOS([0x30, 0x00]);
      } else {
        resp = await NfcManager.transceive([0x30, 0x00]);
      }
      console.warn(resp);

      cleanUp();
    } catch (ex) {
      console.warn("ex", ex);
      cleanUp();
    }
```
Only mifare classic does not work but that is stated in FAQ.

closed 2021-06-28 by whitedogg13

onurmurioglu, 2023-02-14:
I am trying to write and read data into the Desfire card, but I am trying to make a react native application that allows this process to take place in the sector I have determined and that the data on the card cannot be changed by others without my permission, but I have not found a useful solution in this regard. I would be very glad if you could help.