revtel/react-native-nfc-manager #405: Clear NfcManager.getLaunchTagEvent

daheeahn, 2021-06-21:
react-native-nfc-manager: 2.2.1

When the app first starts, the value of getLaunchTagEvent comes out as desired, but I hope it doesn't come out when reloading.
Someone has raised the same issue before, but there is no solution. I hope it gets fixed!

#167

daheeahn, 2021-06-24:
Anyone help?

whitedogg13, 2021-06-25:
@daheeahn Sorry for the late response. By "reloading", do you mean the hot-reload during development?

daheeahn, 2021-06-28:
yes. @whitedogg13

chazmcgrill, 2021-06-28:
I'm finding the getLaunchTagEvent is fixed to what the value when the app is launched from closed status.

For example if I launch the app by pressing the app icon the value is null. This does not change when I subsequently launch by a background NFC read. If I close the app completely and launch by a background tag read the value is set as the NFC data.

whitedogg13, 2021-06-28:
@chazmcgrill this for the issue description, sounds like a bug and I will check it out once I have a moment~

whitedogg13, 2021-07-10:
Hi all, regarding this issue, we have a much better solution in the latest `v3.9.0`. 

In this release, we can retrieve the background tag via `getBackgroundTag` (can be used for the initial tag event as well), and we can also listen to the following tags via `DiscoverBackgroundTag` event. And the `backgroundTag` can be cleared via `clearBackgroundTag` API (though I don't spot a valid use case for this one).

These APIs are available for both iOS and Android.

The whole idea is easier to understand via following code snippet:

```javascript
function onBackgroundTag(bgTag) {
  // bgTag on Android is a complete tag with info such as id, techs
  // bgTag on iOS contains only a `ndefMessage` property
  if (bgTag) {
    navigation.navigate('TagDetail', {tag: bgTag});
  }
}

// get the initial launching tag
const bgTag = await NfcManager.getBackgroundTag();
onBackgroundTag(bgTag);

// listen to other background tags after the app launched
NfcManager.setEventListener(
  NfcEvents.DiscoverBackgroundTag,
  onBackgroundTag,
);
```

A full example app can be found in [NFC ReWriter](https://github.com/revtel/react-native-nfc-rewriter).

I will close it for now, please feel free to re-open!

closed 2021-07-10 by whitedogg13