revtel/react-native-nfc-manager #406: Feature Request: ios background tag

daheeahn, 2021-06-28:
Is there any way to get data from app after tagging iOS background like android's getLaunchEvent?
As mentioned in this [issue](https://github.com/revtel/react-native-nfc-manager/issues/286), is it possible to implement using [this library](https://github.com/FabianGroeger96/NFC-Background-reading)?
I wish it could be implemented.

chazmcgrill, 2021-06-28:
Also looking for this feature, am currently trying to implement the function from [apple docs](https://developer.apple.com/documentation/corenfc/adding_support_for_background_tag_reading) in objective-c but not having much luck. Can launch the app from a background NFC read, would be nice to be able to get the data for the tag as well.

whitedogg13, 2021-06-28:
Hi guys, I think this feature is important as well. I will take a look at how to implement this once I have some spare time~

chazmcgrill, 2021-06-28:
@whitedogg13 much appreciated

daheeahn, 2021-06-29:
The sooner the better, thank you very much :)

whitedogg13, 2021-07-05:
Hi all, I have implemented this feature in the latest `v3.8.0`. Please take a look.

To use this feature, you should first complete the [universal link setup](https://developer.apple.com/library/archive/documentation/General/Conceptual/AppSearch/UniversalLinks.html), otherwise, the background tag reading won't work.

After that, you can integrate related API into your project:

First, call `[NfcManager application:continueUserActivity:restorationHandler:]` from your `AppDelegate.m`:
```objective-c
#import "NfcManager.h"
...
- (BOOL)application:(UIApplication *)application continueUserActivity:(nonnull NSUserActivity *)userActivity
 restorationHandler:(nonnull void (^)(NSArray<id<UIUserActivityRestoring>> * _Nullable))restorationHandler
{
  return [NfcManager application:application
                   continueUserActivity:userActivity
       restorationHandler:restorationHandler];
}
```

Next, in your JS side, call `getBackgroundNdef` to retrieve NDEF scanned from background tag:
```javascript
if (Platform.OS === 'ios') {
  const bgNdef = await NfcManager.getBackgroundNdef();
  console.warn('bgNdef', bgNdef);
}
```

You can refer to our NFC-ReWriter demo app, I have included a [commit](https://github.com/revtel/react-native-nfc-rewriter/commit/0b277b667bbbc1c85d5258a8c7b8b5d5b02eb0a5) to use this latest feature.

Thanks!

fgrs, 2021-07-05:
can't wait to try this out, thank you kindly! @whitedogg13

chazmcgrill, 2021-07-05:
Nice one @whitedogg13 will give it a try as soon as possible.

whitedogg13, 2021-07-10:
Hi all, I have further added some enhancements regarding this feature in latest `v3.9.0`. 

- add a more generic `getBackgroundTag` for both Android / iOS (iOS only reports `Ndef` rather than `Tag`, so in previous version we call it `getBackgroundNdef`)
- add a new `DiscoverBackgroundTag` evet, so the app can listen to background tags after the app is launched

The whole idea is easier to understand via following code snippet:

```javascript
function onBackgroundTag(bgTag) {
  // bgTag on Android is a complete tag with info such as id, techs
  // bgTag on iOS contains only a `ndefMessage` property
  if (bgTag) {
    navigation.navigate('TagDetail', {tag: bgTag});
  }
}

// get the initial launching tag
const bgTag = await NfcManager.getBackgroundTag();
onBackgroundTag(bgTag);

// listen to other background tags after the app launched
NfcManager.setEventListener(
  NfcEvents.DiscoverBackgroundTag,
  onBackgroundTag,
);
```

chazmcgrill, 2021-07-14:
Many thanks, listeners is definitely a good thing. I was using Appstate to check the value when bringing app to the foreground. Theres a chance the value would have not been ready. Am on a different project at the moment. Will give it a try as soon as possible

whitedogg13, 2021-08-04:
Hi all, I will close this issue for now, please re-open or fire another one if there're still problems. Thanks!

freeridre, 2021-11-15:
So, the backgroundTag reading is only available when we the app started, but not terminated?

ChronoByteCosmonaut, 2024-11-21:
> Also looking for this feature, am currently trying to implement the function from [apple docs](https://developer.apple.com/documentation/corenfc/adding_support_for_background_tag_reading) in [...]

This is precisely what I am looking for !!!