revtel/react-native-nfc-manager #407: wrtie plain text onto an empty NfcV tag

Martinchit, 2021-07-03:
How can I write plain text to a nfcV tag? Is there any documentation or example can be shared?

Thanks!

whitedogg13, 2021-07-05:
Hi @Martinchit , you can use `ndefHandler.writeNdefMessage`, sample code looks like this:

```js
    try {
      await NfcManager.requestTechnology(NfcTech.Ndef);
      let bytes = Ndef.encodeMessage([Ndef.textRecord(value)]);
      await NfcManager.ndefHandler.writeNdefMessage(bytes);
    } catch (ex) {
      // handle your exception here
    } finally {
      NfcManager.cancelTechnologyRequest();
    }
```

BTW, your tags need to be properly NDEF-formatted for this to work.

Martinchit, 2021-07-05:
@whitedogg13 Thanks for the help. 

I tried with it and getting the `NFCError:401,nfcd:37` in `ios` and `writeNdefMessage fail` in `android`.

I use `getTag()` and see `"techTypes": ["android.nfc.tech.NfcV", "android.nfc.tech.NdefFormatable"]`. I am not sure if its formatted. If not, how can I format it?

Thanks

whitedogg13, 2021-07-06:
@srluo Do you know which app supports formatting NfcV tags?

srluo, 2021-07-08:
Hi, Richie,

Please download this app:
https://apps.apple.com/tw/app/nfc-tap/id1278913597

After downloaded, launching App and click [Read Tag]. Please check the memory size of this type 5 tag. Or you could directly write the CC files, if this tag was blank.

// writing CC files with magic numbers
Then, switch to [Tag Memory] on bottom of main page. Choose [Write]-> Writing mode:[writeSingleBlock]/ Block Address: [0]/ Block Data: [E1, 40, SS, 00], where SS = total memory size / 8.

// write empty NDEF TLV
Last, same on [Write]-> Writing mode:[writeSingleBlock]/ Block Address: [1]/ Block Data: [03, 00, FE, 00].

Done. Enjoy to write your NDEF now.

Best Regards,
Roger Luo

PS. From my iPhone

closed 2021-09-22 by github-actions[bot]