revtel/react-native-nfc-manager #408: [iOS] Request technology sometimes fails to start.

threequartersjohn, 2021-07-13:
Hi! First, thank you for this package 🙏

Onto the issue, i am trying to implement a functionality by which trying to access an NFC tag will happen within a set timeout, and, in case it fails for some reason before the timeout has finished, it will retry to access the NFC tag. In this implementation, the following steps are taken, after starting the process using `start`:
- Request technology;
- Perform operations;
- In case any of the previous fails, invalidate with error, otherwise invalidate normally;
- In case it was invalidated with error and timeout has not expired, request technology again and restart cycle.

The problem i'm finding is that, after the first time, sometimes requesting technology will never bring up the native NFC modal, effectively not starting the NFC process. This can happen after one retry, or ten retries, even twenty or more, and i'm not able to predict what might be causing it to suddenly not start. Since the timeout eventually ends, i have to exit this process and i'm left with a technology request that never resolved or rejected. 

What happens then, is that any attempt to request technology would error out with "Duplicated registration", even if subsequent `invalidateSession` or `invalidateSessionWithError` calls fulfilled successfully, and, i assume, cleaning up the very sessions that would afterwards cause the "Duplicated registration" error.

From debugging inside the package, the only thing that i can figure out is that for some reason, the failing `requestTechonology` would reach the end of its process, but `tagReaderSessionDidBecomeActive` is never fired. 

What i'm trying to understand is that perhaps there is some asynchronous process that i'm not currently successfully awaiting, because otherwise i'm not sure what can be causing this. I suspect that it might be a race condition, but my unfamiliarity with native code might make me unaware of something else that might be causing this.

I'll continue investigating this issue and will report any advances, but i'm blocked on this bug and it's hard to understand if this is a problem with my implementation or with the package itself, so any help is welcome!

acostalima, 2021-08-10:
@whitedogg13 any idea on what could be happening here? 🙏🏻

threequartersjohn, 2021-08-11:
Some more information on this. It seems i'm more frequently able to replicate this behaviour the quicker i am between retries, that is, it seems like the shorter the time between a session is closed (i.e., the `SessionClosed` event is fired) and a new session is requested through `requestTechnology`, the more frequently this behaviour happens.

Also, this only seems to happen when the use cancels the NFC process through the  `Cancel` button on the native pop-up. In instances where my own application invokes `invalidateSessionWithError`, and the native pop-up has an additional state where it shows the error message with an error icon, i cannot replicate this behaviour at all. This leads me to think that the error state of the native pop-up is critical in some way, perhaps by blocking user interaction for some time as some behind-the-scenes things are finishing up, time that does not exist when the user presses the `Cancel` button since in that instance the pop-up disappears almost instantly and the `SessionClosed` event is fired also almost immediately after. 

I'm still unsure whether this is due to Apple's implementation of the NFC SDK or not. Any help is appreciated.

acostalima, 2021-08-15:
> Some more information on this. It seems i'm more frequently able to replicate this behaviour the quicker i am between retries, that is, it seems like the shorter the time between a session is closed [...]

@whitedogg13 I've put together a [repro](https://github.com/moxystudio/react-native-with-moxy/tree/react-native-nfc-manager-ios-experiment), in isolation from the app we're working on, to demonstrate the issue @threequartersjohn described above. Check out the implementation we've done in the [home screen](https://github.com/moxystudio/react-native-with-moxy/blob/a98b9f2e4211e909f78aa10213b91fa8901f12c8/src/screens/home/HomeScreen.js).

Down below you'll find two videos. When the session is not active, the scan button becomes green. Otherwise, it becomes red. When it becomes blue it means we're pressing on it.

The first video shows the failure scenario where the native NFC scan prompt no longer appears when we call `requestTechnology` for the 2nd time after pressing on the cancel button before on the 1st time. Note that we always wait for the session closed event before making a new call to `requestTechnology`. However, on the 2nd time, the [`tagReaderSessionHasBecomeActive`](https://github.com/revtel/react-native-nfc-manager/blob/484314e4fc8c798791a12f865360bdee512ad5b4/ios/NfcManager.m#L282) callback is never called.

The second video shows the success scenario where a safety timeout starts once the session closed event is fired. This safety timeout ensures the session remains "active" in the app layer for some more time by preventing the user from triggering another scan too early, i.e., before the session has truly stabilised.

https://user-images.githubusercontent.com/2727522/129477289-7be677be-8646-43cf-b188-72f84362af59.MP4

https://user-images.githubusercontent.com/2727522/129477280-c96c3587-b993-4620-b86b-499c3210a1af.MP4

acostalima, 2021-08-15:
Also, I did run a search in the Developer Forum to look for existing questions about this particular issue. I've only found one thread so far more or less related, but it does not exactly related to the behavior we're experiencing here.

- https://developer.apple.com/forums/thread/675452

closed 2021-10-31 by github-actions[bot]

mTRx0, 2022-10-12:
It's been a while but I'm facing this exact issue. Calling `await NfcManager.requestTechnology()` right after a session was invalidated with a `UserCancel` Error starts a new `NFCTagReaderSession` but the delegate methods are never called which results in the `requestTechnology` call being stuck.

@acostalima @threequartersjohn Did you ever find any solution to this or is this just a limitation we have to live with?

threequartersjohn, 2022-10-12:
Hi @mTRx0! This was a long time ago and we're no longer working on this so forgive me if i'm not technical or specific enough. We eventually designed the interaction around this limitation. Our solution was to prevent requesting NFC a given time after we closed the session, e.g.:
- Request NFC;
- Request session closure:
- Block requests  for X seconds (we manually tested values until we found what was best for us);
- App can request NFC again.

Unfortunately this has a UX cost that is very hard to ignore, so there is also that. At the time we found that other apps with NFC for iOS were also using this strategy. We never really found the root cause of this, though it would be interesting to check if something is different with newer iOS versions.

mTRx0, 2022-10-12:
Hey @threequartersjohn, thank you for taking the time to explain your solution!

As far as I understand this seems to either be a bug in CoreNFC or an intentional delay then. I guess we'll have to do the same and design our UX around it.

allenspicer, 2024-08-29:
Below is one additional approach that may help some people. When the user attempts to retry and the resource is not available yet we can invalidate that attempt:

```
const requestTagTechnology = useCallback(async () => {
  try {
    await NfcManager.requestTechnology(SupportedTechs);
  } catch (e) {
    const errorMessage = e instanceof Error ? e.message : String(e);
    if (errorMessage.includes('Duplicated registration')) {
        // Failure due to calling requestTechnology too quickly after a previous call
        // ignore user request
      console.warn('NFC requestTechnology called too quickly after the previous call.');
    } else {
      // Log or report other errors
      console.error('NFC requestTechnology error:', errorMessage);
      throw e;
    }
  }
}, []);
```