revtel/react-native-nfc-manager #409: Set up NFC Manager as a plugin in app.json Expo eas build to write Ndef

freeridre, 2021-07-20:
Hi there!

I try to develop my own IOS NFC app for my access control system.

Expo Completed the NFC integration in development Clients. [EXPO NFC LINK](https://expo.canny.io/feature-requests/p/nfc)
The problem is that I'm not able to write Ndef messages to tags. With **NFC Tools** which an App in Appstore I'm able to write Ndef messages, therefore I think I have to add some permissions to my app, but I don't know how to do that. 
I have this in app.json
```
"plugins": [
      [
        "react-native-nfc-manager",
      ]
    ]
```
Should I do something more in here to be able to write Ndef messages?
When I try to write Ndef Messages the program never reaches 

> console.log("--------------------------------------------");

```
_testNdef = async () => {
      //WritePayload("uniqueId");
      //writeNdef('Hello');
    let result = false;

  try {
    
    // Step 1
    console.log("**********************************************");
    /*await NfcManager.requestTechnology(NfcTech.Ndef, {
      alertMessage: 'Ready to write some NDEF',
    });*/
    await NfcManager.requestTechnology(NfcTech.Ndef);
    console.log("--------------------------------------------");
    let bytes = Ndef.encodeMessage([Ndef.textRecord('Hello NFC')]);
    await NfcManager.ndefHandler.writeNdefMessage(bytes);
    /*if (bytes) {
      await NfcManager.ndefHandler // Step2
        .writeNdefMessage(bytes); // Step3

      if (Platform.OS === 'ios') {
        await NfcManager.setAlertMessageIOS('Successfully write NDEF');
      }
    }*/

    result = true;
  } catch (ex) {
    console.warn(ex);
  }finally{
    NfcManager.cancelTechnologyRequest();
  }

  // Step 4
  NfcManager.cancelTechnologyRequest().catch(() => 0);
  return result;

  }
```
Thank you in advance!

freeridre, 2021-07-20:
Now I tried to set up NFC capabilities, and when I build I got this: 
[image]

freeridre, 2021-07-20:
I got these errors: 
**TagConnectionLost**
```
at [native code]:null in construct
at node_modules\@babel\runtime\helpers\construct.js:22:9 in _construct
at node_modules\@babel\runtime\helpers\wrapNativeSuper.js:26:22 in Wrapper
at [native code]:null in construct
at http://192.168.0.118:8081/index.bundle?platform=ios&dev=true&hot=false&minify=false:142433:321 in _createSuperInternal
at http://192.168.0.118:8081/index.bundle?platform=ios&dev=true&hot=false&minify=false:142444:25 in NfcErrorBase
at [native code]:null in construct
at http://192.168.0.118:8081/index.bundle?platform=ios&dev=true&hot=false&minify=false:142433:321 in _createSuperInternal
at http://192.168.0.118:8081/index.bundle?platform=ios&dev=true&hot=false&minify=false:142549:26 in TagConnectionLost
at node_modules\react-native-nfc-manager\src\NfcError.js:100:26 in buildNfcExceptionIOS
at http://192.168.0.118:8081/index.bundle?platform=ios&dev=true&hot=false&minify=false:142938:38 in handleNativeException$
at node_modules\regenerator-runtime\runtime.js:63:36 in tryCatch
at node_modules\regenerator-runtime\runtime.js:293:29 in invoke
at node_modules\regenerator-runtime\runtime.js:63:36 in tryCatch
at node_modules\regenerator-runtime\runtime.js:154:27 in invoke
at node_modules\regenerator-runtime\runtime.js:166:18 in PromiseImpl.resolve.then$argument_1
at node_modules\react-native\node_modules\promise\setimmediate\core.js:37:13 in tryCallOne
at node_modules\react-native\node_modules\promise\setimmediate\core.js:123:24 in setImmediate$argument_0
at node_modules\react-native\Libraries\Core\Timers\JSTimers.js:130:14 in _callTimer
at node_modules\react-native\Libraries\Core\Timers\JSTimers.js:181:14 in _callImmediatesPass
at node_modules\react-native\Libraries\Core\Timers\JSTimers.js:441:30 in callImmediates
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:387:6 in __callImmediates
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:135:6 in __guard$argument_0
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:364:10 in __guard
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:134:4 in flushedQueue
at [native code]:null in flushedQueue
at [native code]:null in invokeCallbackAndReturnFlushedQueue

```

freeridre, 2021-07-25:
Nobody?

freeridre, 2021-07-26:
@whitedogg13 @poison @dulmandakh @monder @changtimwu Please help me...! 
Here is the expo forum where I commented the latest state of the error:
[https://forums.expo.io/t/eas-build-error-provisioning-profile-com-apple-developer-nfc-readersession-iso7816-select-identifiers/54838/14](url)

Thank you so much!

freeridre, 2021-07-27:
Finally I found the error. 
If you use selectidentifiers as it's described, then the library writes the AIDs into the entitlements which is wrong.
The AIDs have to be in the info.plist not in the entitlements
Like this:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
  <dict>
    <key>CFBundleDisplayName</key>
    <string>Senity</string>
    <key>CFBundleShortVersionString</key>
    <string>1.0.0</string>
    <key>CFBundleURLTypes</key>
    <array>
      <dict>
        <key>CFBundleURLSchemes</key>
        <array>
          <string>com.freeridre.SenityNFCForiOS</string>
        </array>
      </dict>
      <dict>
        <key>CFBundleURLSchemes</key>
        <array>
          <string>exp+senitynfcforios</string>
        </array>
      </dict>
    </array>
    <key>CFBundleVersion</key>
    <string>1</string>
    <key>NFCReaderUsageDescription</key>
    <string>It's for hobby purpose</string>
    <key>UILaunchStoryboardName</key>
    <string>SplashScreen</string>
    <key>UIRequiresFullScreen</key>
    <true/>
    <key>UISupportedInterfaceOrientations</key>
    <array>
      <string>UIInterfaceOrientationPortrait</string>
      <string>UIInterfaceOrientationPortraitUpsideDown</string>
    </array>
    <key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
    <array>
      <string>D2760000850100</string>
      <string>D2760000850101</string>
    </array>
  </dict>
</plist>
```
and the entitlement has to be this:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
  <dict>
    <key>aps-environment</key>
    <string>development</string>
    <key>com.apple.developer.nfc.readersession.formats</key>
    <array>
      <string>NDEF</string>
      <string>TAG</string>
    </array>
  </dict>
</plist>
```

If you use React Native with EAS (Expo Application Services), than you have to write your own info.plist, and  in the app.json like this:
```
"ios": {
      "supportsTablet": true,
      "requireFullScreen": true,
      "bundleIdentifier": "com.freeridre.SenityNFCForiOS",
      "infoPlist": {
        "com.apple.developer.nfc.readersession.iso7816.select-identifiers": [
          "D2760000850100",
          "D2760000850101"
        ]
      }
```
You do not have to specify the entitlement because the react native nfc manager library automatically fills with this, which okey:
```
<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
    <array>
      <string>D2760000850100</string>
      <string>D2760000850101</string>
    </array>
```
I hope I can help with this for others!
@whitedogg13, please update the library. Thank you so much!

closed 2021-07-27 by freeridre

whitedogg13, 2021-07-29:
@freeridre thanks for spotting this, I will check it out~

EvanBacon, 2021-09-15:
@freeridre sorry I didn't see this issue until now, the problem seems to have been resolved in https://github.com/revtel/react-native-nfc-manager/pull/414 sorry for the confusion.