revtel/react-native-nfc-manager #411: want to add pass protection for WRITE operation ....in ntag215/213

vinitchaudhari, 2021-07-23:
i want to protect write operation in ntag215/213.....
but didnt get it how to pass protection ....
after setting password ...then how to authenticate ....then how to rewrite ntag 
i already referred documentation for ntag 213/215 but didnt get that hex code ...
anybody please help...

closed 2021-10-07 by github-actions[bot]

OtayNacef, 2022-03-31:
check this

OtayNacef, 2022-03-31:
> i want to protect write operation in ntag215/213..... but didnt get it how to set pass protection .... after setting password ...then how to authenticate ....then how to rewrite ntag i already referred documentation for ntag 213/215 but didnt get that hex code ... anybody please help...

Hey check this  example https://github.com/whitedogg13/nfc-pokemon-app/blob/main/src/NfcUtils/ensurePasswordProtection.js

vinitchaudhari, 2022-04-02:
can you please explain little bit more about how to set password using
string ?e.g i want to set password "hello world"
thanks in advance