revtel/react-native-nfc-manager #413: Transmitting card data to card reader on touch

GEMI, 2021-07-29:
Hey,
Thanks for creating this library it is really useful. I was wondering how should I make card data available when a user wants to transmit card data when he touches a card reader?

The workflow I want to implement:
1. User touches card to phone, reads card data to phone
2. User touches phone to card reader, card reader picks up the data

How would I implement something like this?
Thanks

gbettinazzi, 2021-08-03:
@GEMI +1. It will be very interesting to do that, even on Android phones only. I'm trying to use this library with no luck
https://github.com/appidea/react-native-hce

sleaper, 2021-09-09:
Did you find any solution? I'm trying to achieve similar thing.

closed 2021-11-24 by github-actions[bot]

freeridre, 2021-11-24:
Well that’s what I implemented. With this library I solved this with NDEF messages. If you use PN532 reader then, it could acts like a virtual writeable NFC tag. You can easily write your card data into it.