revtel/react-native-nfc-manager #415: Can you please provide me with a full example of reading tags ?

Khairala, 2021-08-03:
sorry for asking but am still new in react native but reading NFC is required in the project am working on i've tried alot to read tags and read most of issues and searched YouTube but couldn't find any working example .
so can anyone plz provide me with reading example .

SemihOzdogan, 2021-08-05:
@Khairala 
Yes, I agree with you that the documents are not enough and the usages are not clear. Can you please share a working example for reading and writing ? @whitedogg13

godlatro, 2021-09-16:
better explain is using react-native-nfc-rewriter app
Just run this app and you see how it work

SemihOzdogan, 2021-09-16:
@godlatro  thank you for information. I looked at that application but it couldn't help with some details.For example What is the NdefFormatable ? NdefFormatable convert to Ndef ?

Pinwheeler, 2021-12-16:
+1 on this- the example usage is.... well, kinda bad- it's cut from the example app, and doesn't really work on its own.

Sending people to the example app isn't really ideal either because the example app has a huge amount of (undocumented) behavior and legacy dependencies [reconnect.js](https://www.npmjs.com/package/reconnect)

this repository needs a "simplest possible example" application and/or simplest possible usage. I'm going to try to figure out what that is and make a PR.

closed 2022-04-01 by github-actions[bot]