revtel/react-native-nfc-manager #416: Unsupported Tag Api Error

SemihOzdogan, 2021-08-05:
Hello to everyone.I am getting 'unsupported tag api' error as a result of reading my nfc tag. I want to both read and write to my nfc tag. Can you help?

likrot, 2021-08-05:
What OS do you have? Not all Android devices can read all nfc tags

SemihOzdogan, 2021-08-12:
@likrot  OS Android 8.0.0 version.Reading and writing operations can be done in other nfc applications.

ombogdan, 2021-09-21:
so the problem itself

lukaszrzasa, 2021-09-28:
Same here, I need to erease tag using different application to use writeNdefMessage without error

lukaszrzasa, 2021-09-28:
// Tested with NFC tools

Problem when Technologies available has "NdefFormatable" instead of "Ndef".

After erease tag using "NFC Tools" app, Technologies available has "Ndef" and it's ready to write using this library.

SemihOzdogan, 2021-10-04:
@lukaszrzasa I tried with NFC Tools application but it didn't work. I can't delete the tag. I get a 'Write error' error.

freeridre, 2021-12-04:
Well, I have the same error but only with Android. With Iphone I can easily read-write NDEF messages. So do not really know what's the problem. My Android phone is a Samsung Galaxy s10 plus. @whitedogg13 , do you have any suggestions?

closed 2022-03-20 by github-actions[bot]

HunainHumail, 2022-06-28:
I am having same issue, can we please get the solution for this?
I am not able to write to the nfc in google pixel 3 to google pixel 3, I see that the nfc technologies from google pixel 3 are NfcA and IsoDep, which are supported by your library. But it still still giving api not supported error.
Please help me figure out this problem.. Does anyone know about it?

mTRx0, 2022-10-19:
I'll leave my simple solution to this problem here in case somebody stumbles across this issue.

I was passing  `[NfcTech.Ndef, NfcTech.IsoDep]` to the `requestTechnology()` method since this was working on iOS, but calling the `transceive()` method on android  was throwing this `unsupported tag api` error.
After digging into the native code it seems like you can only use one tech type at a time on android, so either `Ndef` or `IsoDep` (whichever comes first in the array passed to `requestTechnology()` seems to be the one that's used).

Changing the tech array from `[NfcTech.Ndef, NfcTech.IsoDep]` to `[NfcTech.IsoDep, NfcTech.Ndef]` proved to be the solution in my case. Calling the `transceive()` method was working again.
On the other side I'm now unable to write or read Ndef messages.

@whitedogg13 Sorry for bothering you but I'm not quite sure if this is the intended behavior and if it is, is there a workaround / solution? (maybe worth adding to the docs)

shant1010, 2022-12-15:
same issue here. For me ALL NfcTech.Ndef operations fail on my pixel 3a XL (android 12)

Henry-E, 2024-06-25:
Same issue here with reading and writing ndef between a pixel 5 and a samsung galaxy A15