revtel/react-native-nfc-manager #417: Mifare usage Examples

vincenzodomina, 2021-08-07:
Thanks for this great library!

For my use case there was missing some Mifare examples to get started.

This is what is working for me so far, hoping for contribution on more.

Possible README entry:

## Advanced: Mifare Ultralight usage

Here's an example to read a Mifare Ultrlight tag:

```javascript
import NfcManager, {NfcTech} from 'react-native-nfc-manager';

// Pre-step, call this before any NFC operations
async function initNfc() {
  await NfcManager.start();
}

async function readMifare() {
  try {
      // 0. Request Mifare technology
      let reqMifare = await NfcManager.requestTechnology(NfcTech.MifareUltralight);
      if (reqMifare !== 'MifareUltralight') {
          throw new Error('[NFC Read] [ERR] Mifare technology could not be requested');
      };

      // 1. Get NFC Tag information
      const nfcTag = await NfcManager.getTag()
      console.log('[NFC Read] [INFO] Tag: ', nfcTag);

      // 2. Read pages
      const readLength = 60;
      let mifarePages = [];
      const mifarePagesRead = await Promise.all([...Array(readLength).keys()].map(async (_, i) => {
          const pageOffset = i * 4; // 4 Pages are read at once, so offset should be in steps with length 4
          let pages = await NfcManager.mifareUltralightHandlerAndroid.mifareUltralightReadPages(pageOffset);
          mifarePages.push(pages);
          console.log(`[NFC Read] [INFO] Mifare Page: ${pageOffset}`, pages);
          //await wait(500); // If Mifare Chip is to slow
      }));

      // 3. Success
      console.log('[NFC Read] [INFO] Success reading Mifare');

      // 4. Cleanup
      _cleanup();
  } catch (ex) {
      console.warn('[NFC Read] [ERR] Failed Reading Mifare: ', ex);
      _cleanup();
  }
};

function _cleanup() {
    NfcManager.cancelTechnologyRequest().catch(() => 0);
};
```

Would you like pull requests for stuff like this?

whitedogg13, 2021-08-08:
@vincenzodomina Sure, it's awesome! Please fire a PR and I will be very happy to merge. BTW, If you're interested in involving more in this project, please send me an email: whitedogg13@gmail.com :)

vincenzodomina, 2021-10-03:
#441

closed 2021-10-03 by vincenzodomina

gshoanganh, 2022-10-18:
@vincenzodomina  can't use it for mifare classic card.

vincenzodomina, 2022-10-18:
> @vincenzodomina can't use it for mifare classic card.

That is another Nfc technology, right?

w3blogfr, 2023-02-12:
Hello @vincenzodomina , do you may have a write exemple? with authentication?