revtel/react-native-nfc-manager #422: 'writeNdefMessage fail' error with Samsung Galaxy S10 (android 11)

DivyaDevineni, 2021-08-11:
I'm using **Samsung Galaxy S10 running android 11** to write NDEF message to NFC tag (iso15693). My react-native app reads NDEF message successfully but it fails to write back to the tag.

`await NfcManager.writeNdefMessage(bytes)`

Nfc Manager throws error when the above line gets executed 

Error

`writeNdefMessage fail`

Using the other third party apps, my phone is able to write NDEF. So I guess the issue is not with the phone.

Any suggestions would be helpful.

Thanks

srluo, 2021-08-12:
Please refer to:
https://github.com/revtel/react-native-nfc-manager/issues/407

Best Regards,
Roger Luo

PS. From my iPhone

DivyaDevineni, 2021-08-12:
I have also tried using `ndefHandler.writeNdefMessage` and I still get the same error `writeNdefMessage fail`

 ```
try {
      await NfcManager.requestTechnology(NfcTech.Ndef);
      const existingMessage = await NfcManager.getNdefMessage() // can read NDEF message 
      let bytes = Ndef.encodeMessage([Ndef.textRecord(value)]);
      await NfcManager.ndefHandler.writeNdefMessage(bytes); // same error for "NfcManager.writeNdefMessage(bytes)"
    } catch (error) {
      console.log(error) // logged as 'writeNdefMessage fail'
      throw new Error(error)
    } 
```

fails at this line  `await NfcManager.ndefHandler.writeNdefMessage(bytes)`

Note : The tag I am having is NDEF formatted and I'm able to read NDEF message. But having issue to write to tag.

paichato, 2021-08-12:
@DivyaDevineni try this:

```
 async function writeNdef({ type, value }) {

        let result = false;

        try {
            await NfcManager.requestTechnology(NfcTech.Ndef, {
             });
            const info = { `_THE_INFO_YOU_WANT_TO_WRITE`_ };
            const bytes = Ndef.encodeMessage([Ndef.textRecord(JSON.stringify(info))]);
            if (bytes) {
                  await NfcManager.ndefHandler.writeNdefMessage(bytes);
                  }
            result=true;
           catch(ex){
            console.log('Error:'+ex);
           }
       NfcManager.cancelTechnologyRequest().catch(() => 0);
       return result;
       }
```

Then call the function...

closed 2021-10-30 by github-actions[bot]

HunainHumail, 2022-06-28:
I am having same issue, can we please get the solution for this?
I am not able to write to the nfc in google pixel 3 to google pixel 3, I see that the nfc technologies from google pixel 3 are NfcA and IsoDep, which are supported by your library. But it still still giving api not supported error.
Please help me figure out this problem.. Does anyone know about it?

pruthvirajkarurcodewave, 2022-09-12:
Was this resolved ? i am able to write if we just tap it once. But if we keep the nfc tag attached to the phone and then initiates the write process and do it for 2 to 3 times then the below errors will start coming. 

```
1 unsupported tag api
2 Tag is not ndef
3 java.io.IOException
```