revtel/react-native-nfc-manager #423: [Android] Native tag scan window keeps popping up upon reading tag even though NfcManager.start() is earlier called

strid408, 2021-08-12:
Similarily to issue #381, the native android nfc scan keeps popping up upon scanning an nfc tag.

As suggested, I initially call `NfcManager.start()` without errors but still keep getting redirected to the native android tag screen.

On another note, I have no issues reading tags connecting the `readNdef` function to a button, using example code from the documentation.

Device: **Samsung Galaxy S9**
Version: **Android 9**

Abandonedengineer, 2021-09-21:
I'm having the same issue for Samsung A31s any update on this?

Bad-Listener, 2021-11-03:
Hey @whitedogg13, unfortunately the extra declaration on Android Manifest did not solve it.
The only case that is broken is if user disables NFC and then enables it during the lifetime of the app. 
In this case, the first scan is handled by native android. The second tag I scan is always handled by react-native-nfc-manager

Zardozz, 2022-01-12:
Does this happen when the user is using "Quick" settings to turn of NFC not the full setting App? My guess the answer is yes to using the "Quick" settings as this does not pause and resume the App thus the package won't know about the change in status of the NfcAdaptor.

Really the package needs a Broadcaster Receiver to get notified when the state of the NfcAdaptor, gets turned on for it to re-enable it's foreground detection.

ExiledNarwal28, 2022-04-07:
Same issue on Samsung S9+. I can't properly read a tag, the native NFC app keeps popping up

ramzitm91, 2022-06-21:
Hi, any update to this issue? It happened on my several phone test like Redmi Note 8 Pro and Samsung Galaxy A32.
but on Poco F3 Pro and Oneplus 3 seems normal

fpiantoni, 2022-06-23:
Same issue on Samsung S10. I am reading the tag and waiting for the information inside of it, then i am starting a new NfcManager.start() to write on the tag, but when the nfc comes closer the native NFC app keeps popping up. 
Please if anyone knows a solution i am all ears. @whitedogg13

whitedogg13, 2022-06-25:
Hi @fpiantoni , from the description of your issue, I think you might misunderstand our API usage, can you try our demo app to see if it works on your phone? [Google Play link](https://play.google.com/store/apps/details?id=com.washow.nfcopenrewriter)

This app is also written in RN and use our own library, so we can further check what's going wrong. [The app's repo](https://github.com/revtel/react-native-nfc-rewriter)

FYI, I'm testing this app via Samsung Note 10 and it's working properly.

whitedogg13, 2022-06-25:
Hi @Bad-Listener , sorry for the super late response again. 

I understand you found a workaround for the original issue by putting an extra 2500 ms delay, as reported [here](https://github.com/revtel/react-native-nfc-manager/issues/423#issuecomment-954544765). 

But I'm also curious about whether this happened in [our demo app](https://play.google.com/store/apps/details?id=com.washow.nfcopenrewriter), Can you help to test on your phone to see if it happened? Much appreciated!

whitedogg13, 2022-06-25:
@Bad-Listener , I forgot to mention, if the on-off switching during app lifetime is bothering you, you can consider to listen to `NfcEvents.StateChanged` event and disable NFC related feature in your UI accordingly.

For example, that's say we have a React state called `enabled`, which is used to tracking if our device's NFC has been enabled and users are allowed to use NFC related feature only when `enabled` is `true`.

Based on this, we can then listen to `NfcEvents.StateChanged` and change the value for `enabled` state:
```js
          if (Platform.OS === 'android') {
            NfcManager.setEventListener(
              NfcEvents.StateChanged,
              ({state} = {}) => {
                // disable any pending tech requests
                NfcManager.cancelTechnologyRequest().catch(() => 0);
                if (state === 'off') {
                  setEnabled(false);
                } else if (state === 'on') {
                  setEnabled(true);
                }
              },
            );
          }
```

Full example can be found [here](https://github.com/revtel/react-native-nfc-rewriter/blob/master/src/Screens/Home/index.js)

closed 2022-10-10 by github-actions[bot]