revtel/react-native-nfc-manager #427: java.lang.NullPointerException when invoking NfcManager.getTag()

chenghuiyuan, 2021-08-30:
Hi, the app crashes sometimes with the following error message: 

`java.lang.NullPointerException: 
  at community.revteltech.nfc.NfcManager.getTag (NfcManager.java:158)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.facebook.react.bridge.JavaMethodWrapper.invoke (JavaMethodWrapper.java:372)
  at com.facebook.react.bridge.JavaModuleWrapper.invoke (JavaModuleWrapper.java:151)
  at com.facebook.react.bridge.queue.NativeRunnable.run (Native Method)
  at android.os.Handler.handleCallback (Handler.java:883)
  at android.os.Handler.dispatchMessage (Handler.java:100)
  at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage (MessageQueueThreadHandler.java:27)
  at android.os.Looper.loop (Looper.java:237)
  at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run (MessageQueueThreadImpl.java:226)
  at java.lang.Thread.run (Thread.java:919)`

My environment: 
`environment:
 OS:  Android 10
Packages: (installed)
  "@fortawesome/fontawesome-svg-core": "^1.2.34",
    "@fortawesome/free-solid-svg-icons": "^5.15.2",
    "@fortawesome/react-native-fontawesome": "^0.2.6",
    "@react-native-community/async-storage": "^1.12.1",
    "@react-native-community/cli-platform-android": "^4.13.0",
    "@react-native-community/netinfo": "^6.0.0",
    "add": "^2.0.6",
    "apisauce": "^2.0.1",
    "date-fns": "^2.16.1",
    "easy-peasy": "^5.0.3",
    "i18n-js": "^3.8.0",
    "jest": "26",
    "jetifier": "^2.0.0",
    "lodash.memoize": "^4.1.2",
    "lottie-ios": "3.1.8",
    "lottie-react-native": "^4.0.2",
    "native-base": "^2.15.2",
    "react": "17.0.1",
    "react-native": "0.64.2",
    "react-native-countdown-circle-timer": "^2.3.11",
    "react-native-gesture-handler": "^1.10.3",
    "react-native-keychain": "^6.2.0",
    "react-native-localize": "^2.0.3",
    "react-native-modals": "^0.22.3",
    "react-native-nfc-manager": "^3.0.3",
    "react-native-numpad": "https://github.com/chenghuiyuan/react-native-numpad.git",
    "react-native-password-strength-meter": "https://github.com/chenghuiyuan/react-native-password-strength-meter.git",
    "react-native-reanimated": "^1.13.2",
    "react-native-safe-area-context": "^3.1.9",
    "react-native-screens": "^2.17.1",
    "react-native-svg": "^12.1.0",
    "react-native-swipe-list-view": "^3.2.8",
    "react-native-ticker": "^4.0.0",
    "react-navigation": "^4.4.3",
    "react-navigation-tabs": "^2.10.1",
    "yarn": "^1.22.10"`

Thank you!

CYBERKING40, 2021-09-07:
Did you solved the issue?

chenghuiyuan, 2021-09-07:
I used to have this part in my code: 
>       await NfcManager.requestTechnology([NfcTech.Ndef]);
>       tag = await NfcManager.getTag();
>       tag.ndefStatus = await NfcManager.ndefHandler.getNdefStatus();

I think `tag.ndefStatus` causes `java.lang.NullPointerException` when the `Promise` ` await NfcManager.getTag();` is rejected

chenghuiyuan, 2021-09-22:
After modifying to 
> await NfcManager.requestTechnology([NfcTech.Ndef]);
> tag = await NfcManager.getTag();
> if (tag) tag.ndefStatus = await NfcManager.ndefHandler.getNdefStatus();

The crash still happened. Device is Samsung Galaxy A30s - Android 11 (SDK 30)

closed 2021-12-07 by github-actions[bot]