revtel/react-native-nfc-manager #431: [iOS] requestTechnology works, registerTagEvent doesn't

giovanni-caiazzo, 2021-09-09:
Hello,
I have been trying to read from NFC tags with android and iOS. With android there's no problem at all using `registerTagEvent` with all of the NFC that I have, while on iOS if I try to use `registerTagEvent` th NFC popup appears, but then no NFC is read. If, instead, I use `requestTechnology` with tech MifareIOS, my Mifare UltraLight NFC tags are read without problems, but no other non-Mifare type can be read.
Is this an expected behavior or am I missing something?
I have added the `NFCReaderUsageDescription` in `info.plist` and `com.apple.developer.nfc.readersession.formats` in the entitlement file, also xCode has the NFC in capabilities and the app id is registered with nfc as a capability in the developer site.
Here is how I read the NFC in my code for android and iOS:
```import NfcManager, { NfcAdapter, NfcEvents, NfcTech } from 'react-native-nfc-manager';
import { Platform } from 'react-native';
import nfcManager from 'react-native-nfc-manager';

// Pre-step, call this before any NFC operations
export const initNfc = async () => {
    try {
        console.log(await nfcManager.isSupported());
        await NfcManager.start();
        await NfcManager.unregisterTagEvent().catch(() => 0);
        if (Platform.OS !== 'ios') {
            await NfcManager.registerTagEvent({
                isReaderModeEnabled: true,
                readerModeFlags: NfcAdapter.FLAG_READER_NFC_A | NfcAdapter.FLAG_READER_NFC_B | NfcAdapter.FLAG_READER_NFC_F | NfcAdapter.FLAG_READER_NFC_V,
            });
        }
    } catch (err) {
        console.log(err);
    }
    console.log('NFC manager started!');
};

export const StopListeningToNFC = async () => {
    await NfcManager.unregisterTagEvent().catch(() => 0);
    if (Platform.OS !== 'ios') {
        await NfcManager.registerTagEvent({
            isReaderModeEnabled: true,
            readerModeFlags: NfcAdapter.FLAG_READER_NFC_A | NfcAdapter.FLAG_READER_NFC_B | NfcAdapter.FLAG_READER_NFC_F | NfcAdapter.FLAG_READER_NFC_V,
        });
    }
    NfcManager.setEventListener(NfcEvents.DiscoverTag, null);
    NfcManager.setEventListener(NfcEvents.SessionClosed, null);
};

export const StartListeningToNFC = async (route, navigation) => {
    console.log('Started listening to NFC');
    NfcManager.setEventListener(NfcEvents.DiscoverTag, tag => {
        console.log('Found tag! Info is:', tag);
        navigation.setParams({ qrData: tag.id, source: 'nfc', previousPage: route?.params?.previousPage, closed_nfc_session: false });
        StopListeningToNFC().then();
    });
    if (Platform.OS !== 'ios') {
        await NfcManager.unregisterTagEvent().catch(() => 0);
        await NfcManager.registerTagEvent({
            isReaderModeEnabled: true,
            readerModeFlags: NfcAdapter.FLAG_READER_NFC_A | NfcAdapter.FLAG_READER_NFC_B | NfcAdapter.FLAG_READER_NFC_F | NfcAdapter.FLAG_READER_NFC_V,
        });
    } else {
        await _test();
    }
    NfcManager.setEventListener(NfcEvents.SessionClosed, () => {
        console.log('Session closed');
        navigation.setParams({ ...route.params, closed_nfc_session: true });
        StopListeningToNFC().then();
    });
};

const _cleanUp = () => {
    NfcManager.cancelTechnologyRequest().catch(() => 0);
};

const _test = async () => {
    try {
        let tech = Platform.OS === 'ios' ? NfcTech.MifareIOS : NfcTech.NfcA;
        let resp = await NfcManager.requestTechnology(tech, {
            alertMessage: 'Ready to do some custom Mifare cmd!',
        });
        console.warn(resp);

        // the NFC uid can be found in tag.id
        let tag = await NfcManager.getTag();
        console.warn(tag);

        if (Platform.OS === 'ios') {
            resp = await NfcManager.sendMifareCommandIOS([0x30, 0x00]);
        } else {
            resp = await NfcManager.transceive([0x30, 0x00]);
        }
        console.warn(resp);

        _cleanUp();
    } catch (ex) {
        console.warn('ex', ex);
        _cleanUp();
    }
};
```
My `package.json` has the following dependencies:
```
"dependencies": {
    "@react-native-async-storage/async-storage": "~1.15.0",
    "@react-native-community/masked-view": "0.1.10",
    "@react-native-community/netinfo": "6.0.0",
    "@react-navigation/native": "^5.9.4",
    "@react-navigation/stack": "^5.14.5",
    "amazon-cognito-identity-js": "^5.0.3",
    "aws-amplify": "^4.1.2",
    "aws-amplify-react": "^5.0.4",
    "aws-amplify-react-native": "^5.0.2",
    "date-fns": "^2.22.1",
    "expo": "^42.0.0",
    "expo-barcode-scanner": "~10.2.2",
    "expo-constants": "~11.0.1",
    "expo-font": "~9.2.1",
    "expo-location": "~12.1.2",
    "expo-splash-screen": "~0.11.2",
    "expo-status-bar": "~1.0.4",
    "expo-updates": "~0.8.0",
    "react": "17.0.1",
    "react-dom": "17.0.1",
    "react-native": "0.64.2",
    "react-native-gesture-handler": "~1.10.2",
    "react-native-nfc-manager": "^3.10.0",
    "react-native-paper": "^4.9.2",
    "react-native-picker-select": "^8.0.4",
    "react-native-reanimated": "~2.2.0",
    "react-native-safe-area-context": "3.2.0",
    "react-native-screens": "~3.4.0",
    "react-native-unimodules": "~0.14.3",
    "react-native-vector-icons": "^8.1.0",
    "react-native-web": "~0.13.12"
  }
```

msoose, 2021-09-11:
What type of tag technology are you using? I seem to be running in to this issue with an iso15693 tag. Android detects it just find and provides the UID, but iOS does not. 

Entitlements are set correctly, I can read a tag, I just never get a UID returned. Strangely, the example app does not use ios15693 handler at all and it is able to read the UID.

giovanni-caiazzo, 2021-09-13:
One tag has types ["android.nfc.tech.NfcA", "android.nfc.tech.MifareUltralight", "android.nfc.tech.Ndef"],
the other  ["android.nfc.tech.IsoDep", "android.nfc.tech.NfcA"].
If I use registerTagEvent on iOS they are not read or detected at all, while android can read them without problems. If I use requestTechnology using NfcTech.MifareIOS (or NfcTech.NfcA) then the mifare one is detected correctly by iOS, while the other is not.

closed 2021-11-29 by github-actions[bot]

flyskywhy, 2026-03-02:
System NFC alert modal can't show on my Android phone, so be this wrapper [react-native-nfc-rw](https://github.com/flyskywhy/react-native-nfc-rw?tab=readme-ov-file) to read and write NfcV on iOS and Android , maybe you can ref to it.