revtel/react-native-nfc-manager #432: ISO15693 Detection

msoose, 2021-09-10:
Is it necessary to declare anything before attempting to read an ISO15693 tag to indicate the chip to expect an ISO15693 tag on iOS? I haven't needed to 'set' a specific technology on Android and my tag is detected and read just fine, but it doesn't detect it on an iOS device.

I'm using NfcManager.setEventListener(NfcEvents.DiscoverTag, (tag) => {

Thanks in advance

msoose, 2021-09-10:
Looks like I should be using NfcManager.requestTechnology() for this for iOS

I still can't seem to get a UID though...

let iostech = NfcTech.Iso15693IOS;
let iosnfc = NfcManager.requestTechnology(iostech, {alertMessage: 'Tap to scan'})
      console.log('[+] iOS:' + iosnfc)
      let tag = NfcManager.getTag();
      console.log('[+] UID:' + tag.id);

Returns undefined for tag.

msoose, 2021-09-10:
Returning tag stringify I get _U:0, _V,0, _W:null and X:null.

closed 2021-09-12 by msoose

gregogalante, 2021-09-17:
@msoose I'm getting the same problem. How do you solve this?

msoose, 2021-09-17:
I used requestTechnology() on iOS and stuck with setEventListener() for Android devices.

msoose, 2021-09-17:
> @msoose I'm getting the same problem. How do you solve this?

 I used requestTechnology() on iOS and stuck with setEventListener() for Android devices.

Sixzero, 2021-11-02:
So the iOS is working with requestechnology being set?
For android I requetested technology: 
NfcManager.requestTechnology(NfcTech.NfcV)