revtel/react-native-nfc-manager #433: App crash - switching from foreground to background and vice versa

ognjenmarcheta, 2021-09-13:
Android: Production app crashes with the following error when user puts the app in the background and in the foreground again. Did I miss something in the setup?

`FATAL EXCEPTION: mqt_native_modules

java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String[] android.nfc.Tag.getTechList()' on a null object reference

FATAL EXCEPTION: mqt_native_modules
Process: com.payuca.nfc, PID: 32690
java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String[] android.nfc.Tag.getTechList()' on a null object reference
	at community.revteltech.nfc.NfcManager.getTag(NfcManager.java:167)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
	at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:151)
	at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
	at android.os.Looper.loop(Looper.java:246)
	at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:226)
	at java.lang.Thread.run(Thread.java:923)`

Bad-Listener, 2021-09-14:
Having the same issue on Android, also happens on development mode

ognjenmarcheta, 2021-09-15:
I've fixed it by adding an additional check if the tag is not null and it seems that it is working.

NfcManager.java

`@ReactMethod
    public void getTag(Callback callback) {
        synchronized(this) {
            if (techRequest != null) {
                Tag tag = techRequest.getTagHandle();
                if(tag != null){
                WritableMap parsed = tag2React(tag);
                if (Arrays.asList(tag.getTechList()).contains(Ndef.class.getName())) {
                    try {
                        Ndef ndef = Ndef.get(tag);
                        parsed = ndef2React(ndef, new NdefMessage[] { ndef.getCachedNdefMessage() });
                    } catch (Exception ex) {
                        // bypass
                    }
                }
                callback.invoke(null, parsed);
            }else{
                callback.invoke(ERR_NO_TECH_REQ); //This is our crash I've just reused same error message string.
            }
            } else {
                callback.invoke(ERR_NO_TECH_REQ);
            }
        }
    }`

buekera, 2021-09-17:
Hey there, I'm having the same issue.

Steps to reproduce:

1. Call scanner to open so it awaits card/chip to be held to the phone
2. Press "home" button or swipe so app goes into background
3. Reopen app by hitting the app icon again

Now you will be greeted by `Attempt to invoke virtual method 'java.lang.String[] android.nfc.Tag.getTechList()' on a null object reference` as stated by @ognjenmarcheta above.

@ognjenmarcheta is it possible for you to create a pull request to fix the issue within this repository?

Thanks

ognjenmarcheta, 2021-09-21:
@buekera 
Here it is: https://github.com/revtel/react-native-nfc-manager/pull/438

closed 2021-09-21 by ognjenmarcheta