revtel/react-native-nfc-manager #437: AppState background triggered while scanning nfc chip

buekera, 2021-09-17:
Hey there, I just noticed some odd behavior while scanning nfc chips.
The moment you put your nfc chip to the phone and the scanner detects it the AppState changes from `active` to `background` back to `active` again.

Due to this behavior some other parts which listen to the AppStage change behave strangely. Is there any way to work around this or might this even be a bug?

OS: Android
tested nfc manager versions: 3.7.0 - 3.10.0

Code:
```javascript
try {
  setWaiting(true);
  await NfcManager.requestTechnology([NfcTech.Ndef]);

  const tag = await NfcManager.getTag();

  handleNfcScan(tag?.id as unknown as string);
  NfcManager.cancelTechnologyRequest();
} catch (error) {
  // handle error
}
```

Regards

ognjenmarcheta, 2021-09-21:
@buekera That is happening because of how react native sees/defines background and foreground states.
React native uses your app activity as the owner of the UI thread. When something is in front of your app (alerts, dialogs, file pickers, etc.) onPause callback will be called and trigger "background" event.

I've already recently reported this to react-native community repo and it was already reported couple years ago, I think in 2018 but old report was closed due to inactivity.

whitedogg13, 2021-09-21:
@buekera might relate to #438  please try the latest `v3.10.1`. Thanks!

closed 2021-12-22 by github-actions[bot]

neiracausevic1, 2024-01-05:
I fixed it similarly to the solution provided above but just set the difference to 500 ms because it usually takes 20-40 ms if its the NFC triggering the state changes and it works for me.

```
const appState = useRef(AppState.current);
const [lastBackgroundTime, setLastBackgroundTime] = useState(null);

useEffect(() => {
    const handleAppStateChange = nextAppState => {
      if (
        appState.current.match(/inactive|background/) &&
        nextAppState === ‘active’
      ) {
        const currentTime = new Date().getTime();
        if (lastBackgroundTime && currentTime - lastBackgroundTime > 500) {
           // App was in background
        } else {
          // App state falsely triggered by NFC
          return;
        }
      }
      if (nextAppState === ‘background’) {
        setLastBackgroundTime(new Date().getTime());
      }
      appState.current = nextAppState;
    };
    // Subscribe to app state changes
    const appStateSubscription = AppState.addEventListener(
      ‘change’,
      handleAppStateChange,
    );
    return () => {
      // Clean up the subscription on unmount
      appStateSubscription.remove();
    };
  }, [lastBackgroundTime]);

```

douglasjunior, 2024-05-03:
On Android, the `background` state means that the React Native Activity is in background, and not necessary the entire app.

To handle this, I created a package that implements the Android Lifecycle API for React Native: https://github.com/douglasjunior/react-native-applifecycle

> ## Why Use This?
> The original AppState API provided by React Native behaves differently between Android and iOS, particularly regarding the background state:
>
> - On iOS, the `background` state signifies that the entire app is in the background.
> - On Android, the `background` state indicates that the React Native Activity is in the background, which might not necessarily mean the entire app is in the background.
> 
> By using `react-native-applifecycle`, you can handle these differences seamlessly across both platforms, ensuring that the state `background` will be dispatched only when the entire app is in background.