revtel/react-native-nfc-manager #442: How do I change 'NdefFormatable' type to 'Ndef' technology type?

SemihOzdogan, 2021-10-04:
The technology types of my nfc tag are 'NfcV' and 'NdefFormatable'. I can't read and write to my nfc tag.
**Note**: I can read and write to my tag whose technology is 'NfcV' and 'Ndef'. I think I can't read and write because this is 'NdefFormatable'

krishnabojha, 2021-11-10:
I am facing the same issue.
I am not able to read/ write NFC cards because initially, the NFC cards are  'NfcV' and 'NdefFormatable'. Once I write data from the NFC tool then it works fine.
please help me

xgrimaldi, 2022-01-13:
@SemihOzdogan On my side, i was able to read ['NfcV','NdefFormatable']. 
`await NfcManager.requestTechnology([NfcTech.Ndef,NfcTech.NfcV])`

But i'm not able to format 'NdefFormatable' to 'Ndef'.

I've seen in the demo app "React Native NFC ReWriter App", a method in Tool kit that should be able to erase content and format 'NfcA' to 'Ndef'( [LINK to CODE](https://github.com/revtel/react-native-nfc-rewriter/blob/ed10c92b28989ed410521d2590f30f3a783bb4a3/src/NfcProxy.js#L238))
Unfortunately, it's not working in our case to pass from ['NfcV','NdefFormatable'] to '['NfcV','Ndef']'. 

With an older version (^1.2.2 in package.json), i was able to use the **requestNdefWrite** function, with options 'format' or 'formatReadOnly' that help me to format from ['NfcV','NdefFormatable'] to '[Ndef]'. I've updated today to the new version and refactorize my app code, and i've seen that a lot of things have changed, especially how we can format NFC tag. 

@whitedogg13 Is it possible to know the reason why we can't format like in the past ? Or have you got any idea how to format to Ndef nowadays with this version (3.11.5) ? 

Many thanks by the way for your work.
I'll try to go deeper in your code to understand, and see how we can proceed.

whitedogg13, 2022-01-15:
@xgrimaldi hi, thanks for reporting this. You’re right, the formatting API for Android has been missing during my V3 rewrite. I will bring it back soon!

xgrimaldi, 2022-01-17:
Great news.  
I needed to have two version (V1 and V3) to be able to do it actually.. so all in the same version would be great. 😄 
Thanks a lot, i'm waiting for your update.

whitedogg13, 2022-01-19:
Hi @xgrimaldi `v3.12.0` released, which includes the `NdefFormatable` NfcTech for Android. You can use it like this:

```javascript
    try {
      await NfcManager.requestTechnology([NfcTech.NdefFormatable]);
      const bytes = Ndef.encodeMessage([Ndef.textRecord('hello nfc')]);
      await NfcManager.ndefFormatableHandlerAndroid.formatNdef(bytes, {readOnly: true});
    } catch (ex) {
      console.warn(ex);
      handleException(ex);
    } finally {
      NfcManager.cancelTechnologyRequest();
    }
```

Thanks!

closed 2022-01-19 by whitedogg13

xgrimaldi, 2022-01-19:
> Hi @xgrimaldi `v3.12.0` released, which includes the `NdefFormatable` NfcTech for Android. You can use it like this:
> 
> ```js
>     try {
>       await NfcManager.requestTechnology([NfcTech.NdefFormatable]);
>       const bytes = Ndef.encodeMessage([Ndef.textRecord('hello nfc')]);
>       await NfcManager.ndefFormatableHandlerAndroid.formatNdef(bytes, {readOnly: true});
>     } catch (ex) {
>       console.warn(ex);
>       handleException(ex);
>     } finally {
>       NfcManager.cancelTechnologyRequest();
>     }
> ```
> 
> Thanks!

Hi @whitedogg13.
Thank you very much for the quick update, and for your work @whitedogg13. Much appreciated. 👏 😃 
I'll test it tonight, and i'll give you some feedback.

barimehdi77, 2023-09-26:
> ```js
> await NfcManager.requestTechnology([NfcTech.NdefFormatable]);
>       const bytes = Ndef.encodeMessage([Ndef.textRecord('hello nfc')]);
>       await NfcManager.ndefFormatableHandlerAndroid.formatNdef(bytes, {readOnly: true});
> ```

hello @whitedogg13 I tried to use this to write some data on a NFC tag with the following types 
`"techTypes": ["android.nfc.tech.NfcA", "android.nfc.tech.MifareUltralight", "android.nfc.tech.NdefFormatable"]`

the problem is I get an error `[Error: java.io.IOException]`

is there any thing I'm missing out?

Antunsvetic, 2023-12-05:
@whitedogg13  Richie, please help us. I have same problem as @barimehdi77. If card is formatted i can read and write ndef messages that is fine. But formatting is throwing java.io.IOException when formating card to Ndef using ndefFormatableHandlerAndroid.