revtel/react-native-nfc-manager #455: [Question] Make sure NFC is turned off

aymkin, 2021-11-19:
In my React Native app, one of the screens shows the QR code. That QR code should be placed to the QR code reader, which has also NFC capabilities.

My task is to make sure that the NFC is turned off. Yep, my app does not have any NFC features or capabilities. No permissions are set for NFC in AndroidManifest or Info.plist.

That is just a QR code provider requirement to make sure the NFC is off. How to make it?

Is that possible that another app somehow uses NFC at this time?

srluo, 2021-11-19:
Hi, keep QRcode scanning over 10cm distances, you phone will never trigger to NFC scanning!

Best Regards,
Roger Luo

PS. From my iPhone

aymkin, 2021-11-19:
> Hi, keep QRcode scanning over 10cm distances, your phone will never trigger NFC scanning!
> […](#)

Not sure I can teach a thousand users to keep iPhones not closer than 10 cm.

aymkin, 2021-11-19:
Generally, the issue is with Apple Pay and Google Pay. While trying to scan the QR code from the screen both payment applications are popped up, which prevents reading the QR code.

So the idea is just to disable those payment applications while using our app.

- for iOS is possible to use entitlement https://developer.apple.com/documentation/passkit/pkpasslibrary/1617078-requestautomaticpasspresentation

- did not find something similar for Android yet.

closed 2022-03-05 by github-actions[bot]

BhavyaKoshiya, 2023-08-04:
@aymkin 
 Did you manage to add this in the app?

aymkin, 2023-08-06:
> @aymkin 
>  Did you manage to add this in the app?
> 
> 

Yes, we did it

BhavyaKoshiya, 2023-08-06:
> 

Can you provide detailed instruction or a piece of code or something that can guide me through how can i do it with react native?

fikkatra, 2024-04-04:
Try adding this to the `didFinishLaunchingWithOptions` in `AppDelegate.m`:

```
if ([PKPassLibrary isPassLibraryAvailable]) {
    [PKPassLibrary requestAutomaticPassPresentationSuppressionWithResponseHandler:^(PKAutomaticPassPresentationSuppressionResult result) {
        if (result == PKAutomaticPassPresentationSuppressionResultSuccess) {
            RCTLogInfo(@"Apple Pay Automatic Pass Presentation suppressed");
        } else {
            RCTLogInfo(@"Could not suppress Apple Pay Automatic Pass Presentation");
        }
    }];
}
```

This code won't work without adding the correct entitlement. In the entitlement’s file of the target in Xcode, add the `com.apple.developer.passkit.pass-presentation-suppression` key with a Boolean value of YES. You need permission from Apple to include this entitlement.