revtel/react-native-nfc-manager #460: How to open my app

freeridre, 2021-12-01:
Hi there!
So, when my app is closed, and I read a specific NDEF message from the tag, I would like to open my app by NDEF message of the tag, even if my app is in background.
How could I implement this?

Langsdorf, 2021-12-02:
Try using [deep linking](https://blog.logrocket.com/understanding-deep-linking-in-react-native/)

freeridre, 2021-12-03:
> Try using [deep linking](https://blog.logrocket.com/understanding-deep-linking-in-react-native/)

When my app is not running? Should I have to implement a service? what about Headless JS? And I use expo managed workflow.

Langsdorf, 2021-12-03:
In the post I mentioned, it says you can use even if  your app is closed (the link will open it up)

[image]

likrot, 2021-12-24:
> > Try using [deep linking](https://blog.logrocket.com/understanding-deep-linking-in-react-native/)
> 
> When my app is not running? Should I have to implement a service? what about Headless JS? And I use expo managed workflow.

@freeridre You can write such information on NFC chip (with [this](https://play.google.com/store/apps/details?id=com.wakdev.wdnfc&hl=en_US&gl=US) app for instance). 
When scanning with NFC, your app will be open (if it's been installed).

I'm not sure about expo but it works with bare RN app or with an app written with Kotlin so I think this is a system thing.

closed 2022-04-09 by github-actions[bot]