revtel/react-native-nfc-manager #469: Execution failed for task ':react-native-nfc-manager:compileDebugJavaWithJavac'.

Monisha-Raju, 2021-12-13:
[image]
cannot find symbol 'FLAG_MUTABLE' with the version v3.11.1

whitedogg13, 2021-12-14:
@Monisha-Raju thanks for spotting this!

To fix this, please update your `compileSdkVersion` in `android/build.gradle`, for example:
```
buildscript {
    ext {
        ...
        compileSdkVersion = 31
        ...
    }
    ...
}
```

The reason for this is because Android puts new limitation on [PendingIntent](https://developer.android.com/reference/android/app/PendingIntent#FLAG_MUTABLE):

> Starting with Build.VERSION_CODES.S, it will be required to explicitly specify the mutability of PendingIntents

So we have to do this in-order to support Android 12.

Monisha-Raju, 2021-12-21:
> @Monisha-Raju thanks for spotting this!
>
> To fix this, please update your `compileSdkVersion` in `android/build.gradle`, for example:
>
> ```
> buildscript {
>     ext {
>         ... [...]

> @Monisha-Raju thanks for spotting this!
>
> To fix this, please update your `compileSdkVersion` in `android/build.gradle`, for example:
>
> ```
> buildscript {
>     ext {
>         ... [...]

@whitedogg13 - Changing  compileSdkVersion to 31 resolved the 'FLAG_MUTABLE' issue, But it is throwing error in the java compiler. The source and target compatibility is JavaVersion.VERSION_1_8. Do you have any suggestion for this issue?

[image]

hariantara, 2021-12-22:
I have fix on my local and no need to update buildscript on `/android/build.gradle`

`
buildscript {
    ext {
        buildToolsVersion = "30.0.2"
        minSdkVersion = 21
        compileSdkVersion = 30
        targetSdkVersion = 30
        ndkVersion = "21.4.7075529"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:4.2.2")
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}
`

## Step to Fix this issue:

* Open Project (only android folder) on Android Studio
* Go to `react-native-nfc-manager/java/community.revteltech.nfc/NfcManager.java`
*  after that try to change `NfcManager.java` code into this 

`
class NfcManager extends ReactContextBaseJavaModule implements ActivityEventListener, LifecycleEventListener {
    private static final String LOG_TAG = "ReactNativeNfcManager";
    private final List<IntentFilter> intentFilters = new ArrayList<IntentFilter>();
    private final ArrayList<String[]> techLists = new ArrayList<String[]>();
    ...........
    private int FLAG_MUTABLE;  // -----------------> add this

    class WriteNdefRequest {
        NdefMessage message;

    ................

  private PendingIntent getPendingIntent() {
        Activity activity = getCurrentActivity();
        Intent intent = new Intent(activity, activity.getClass());
        intent.addFlags(Intent.FLAG_ACTIVITY_SINGLE_TOP | Intent.FLAG_ACTIVITY_CLEAR_TOP);

        int flag = 0;
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
          flag = FLAG_MUTABLE;  // -----------------------------> add this
        }
        return PendingIntent.getActivity(activity, 0, intent, flag);
    }
`

* after update that file, try to Invalidate Caches/Restart on Android Studio at menu file, and then Clean Build and Re Build again

IsTheJack, 2021-12-22:
@whitedogg13 Same issue on Expo as well. How to figure it out?

whitedogg13, 2022-01-15:
the same as #473, close

closed 2022-01-15 by whitedogg13

ycdaskin, 2022-04-06:
i made compileSdkVersion = 31 but this time firebase crashlytics fired errors while building. so i solved it like this:

1. fork the project
2. go to /react-native-nfc-manager/android/src/main/java/community/revteltech/nfc/NfcManager.java file
3. go to line 1213 and change this section to this:

` 
private PendingIntent getPendingIntent() {
        Activity activity = getCurrentActivity();
        Intent intent = new Intent(activity, activity.getClass());
        intent.addFlags(Intent.FLAG_ACTIVITY_SINGLE_TOP | Intent.FLAG_ACTIVITY_CLEAR_TOP);

        int flag = 0;
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
            flag = 33554432; //PendingIntent.FLAG_MUTABLE;  
        }
        return PendingIntent.getActivity(activity, 0, intent, flag);
    }
`

use your version of the package. clean gradle cache and build the project.

vimosan7, 2023-03-03:
Android 12/API 31 was giving me trouble, downgrading react-native-nfc-manager to v3.14.1 resolved the issue for me