revtel/react-native-nfc-manager #473: Expo support is down

IsTheJack, 2021-12-22:
Versions:
- Node: 16.13.0
- Expo: 44.0.0
- Eas: eas-cli/0.42.4
- System: wsl-x64 Ubuntu 18.04.4 LTS

That issue is related to https://github.com/revtel/react-native-nfc-manager/issues/469, but maybe it have a different solution

I'm getting this error on EAS Build:
```
[stderr]           flag = PendingIntent.FLAG_MUTABLE;
[stderr]                               ^
[stderr]   symbol:   variable FLAG_MUTABLE
[stderr]   location: class android.app.PendingIntent
[stderr] Note: /root/workingdir/build/whitelabel-app/node_modules/react-native-nfc-manager/android/src/main/java/community/revteltech/nfc/NfcManager.java uses or overrides a deprecated API.
[stderr] Note: Recompile with -Xlint:deprecation for details.
[stderr] 1 error
[stderr] FAILURE: Build failed with an exception.
[stderr] * What went wrong:
[stderr] Execution failed for task ':react-native-nfc-manager:compileReleaseJavaWithJavac'.
[stderr] > Compilation failed; see the compiler error output for details.
[stderr] * Try:
[stderr] Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
[stderr] * Get more help at https://help.gradle.org
[stderr] BUILD FAILED in 5m 25s
```

I tried to solve that adding `"minSdkVersion": 31` to  `"react-native-nfc-manager"` plugin config on `app.json`:

```json
{
    ...
    "plugins": [
      [
        "react-native-nfc-manager",
        {
          "nfcPermission": "This app uses the nfc to scan customer codes",
          "compileSdkVersion": 31 <- add this to your config
        }
      ]
    ],
    ...
}
```

But I keep getting the same error.

whitedogg13, 2021-12-23:
I think the easiest solution for now is to use `react-native-nfc-manager@3.11.0`

I’m not familiar with Expo, but from some [SO discussion](), it seems that you still need to modify `android/build.gradle` to change the `compileSdkVersion`.

Even if you have changed this, this might still not work because it still depends on whether the build server have installed the required toolchain for android sdk 31.

whitedogg13, 2021-12-23:
Call out to the expert! @EvanBacon hi, do you know whether EAS build supports `compileSdkVersion` 31 (for Android 12) now? 

Some native breaking change forced the app to be compiled with android sdk 31 to support Android 12:
- [Starting with Build.VERSION_CODES.S, it will be required to explicitly specify the mutability of PendingIntents](https://developer.android.com/reference/android/app/PendingIntent#FLAG_MUTABLE)

Thanks!

IsTheJack, 2021-12-23:
Thanks, @whitedogg13. With that version, the android app worked here. I'll keep following this thread and keep searching for solutions on the Expo side.

Cheers!

rossjcooper, 2022-01-12:
I'm also getting the same issue using expo, if I find a solution I'll post it here

rossjcooper, 2022-01-12:
It may not be ideal, but running `expo prebuild` fixed this for me, obviously be cautious when running this if you have modified your `ios/` or `android/` files as [explained here](https://github.com/expo/fyi/blob/main/prebuilding.md).

EvanBacon, 2022-01-12:
@IsTheJack `compileSdkVersion` is not a [property on the config plugin](https://github.com/revtel/react-native-nfc-manager#props) so it won't do anything.

closed 2022-01-13 by whitedogg13

ycdaskin, 2022-04-06:
making "minSdkVersion": 31 seems to work but this time firebase crashlytics fires an error while building. Is there any way to arrange different "minSdkVersion"s for different packages?

IsTheJack, 2022-06-15:
To fix that on Expo apps. Get the last version of `react-native-nfc-manager`, install `expo-build-properties` (`expo install expo-build-properties`). Finally, add it to  `app.json`: 

```json
{
  "expo": {
    ...
    "plugins": [
      [
        "expo-build-properties",
        {
          "android": {
            "compileSdkVersion": 31
          }
        }
      ]
    ]
  }
}
```