revtel/react-native-nfc-manager #476: TagEvent id is string instead of number[]

raydenvoldeskine, 2022-01-03:
react-native-nfc-manager: 3.1.0

Running on Android (do not know if the problem exist on iOS yet).
Reading tag in TypeScript using NfcManager.getTag() 

EXPECTED:  As the TagEvent id is defined like this in index.d.ts 

```
  export interface TagEvent {
    ndefMessage: NdefRecord[];
    maxSize?: number;
    type?: string;
    techTypes?: string[];
    id?: number[];
  }
```
I expect it to be a number array.

OBSERVED: 

However if I dump the attribute tyoe with typeof it is "string"

whitedogg13, 2022-01-06:
Hi @raydenvoldeskine your observation is correct! The typing is wrong, I will fix it soon.

whitedogg13, 2022-01-09:
Fixed in `v3.11.3`

closed 2022-01-09 by whitedogg13

raydenvoldeskine, 2022-01-09:
Is the change sufficient ? I mean the issue happens "between the worlds"  - conversion from native to type script, would simple change of type script type solve it?

whitedogg13, 2022-01-09:
@raydenvoldeskine yes, it’s just a typing issue and native side will always set the `.id` field to string if it presents.