revtel/react-native-nfc-manager #480: Android OS 12 issues

Waltari10, 2022-01-14:
We are using this Library to scan Abbott Continous Blood Glucose monitoring devices.

Currently we are facing a lot of issues with Android OS 12. A lot of customers are reaching out to us that the scanning is failing and only on OS 12.

One of the users is receiving errors like:
- Error: transceive fail
- Error: no tech request available
- Error: You can only issue one request at a time
- And some of these errors don't have error message since the library is fond of throwing empty errors

Some stacktraces from the same user
```
Error: No error message
  at Error([native code])
  at _construct(node_modules/@babel/runtime/helpers/construct.js:12:53)
  at _construct(node_modules/@babel/runtime/helpers/construct.js:19:21)
  at Wrapper(node_modules/@babel/runtime/helpers/wrapNativeSuper.js:26:14)
  at NfcErrorBase(node_modules/react-native-nfc-manager/src/NfcError.js:1:1)
  at n(node_modules/react-native-nfc-manager/src/NfcError.js:3:14)
  at UserCancel(node_modules/react-native-nfc-manager/src/NfcError.js:1:1)
  at UserCancel(node_modules/react-native-nfc-manager/src/NfcError.js:18:14)
  at buildNfcExceptionAndroid(node_modules/react-native-nfc-manager/src/NfcError.js:139:16)
  at call(node_modules/react-native-nfc-manager/src/NfcManagerAndroid.js:39:13)
  at tryCatch(node_modules/regenerator-runtime/runtime.js:63:40)
  at call(node_modules/regenerator-runtime/runtime.js:294:22)
  at tryCatch(node_modules/regenerator-runtime/runtime.js:63:40)
  at invoke(node_modules/regenerator-runtime/runtime.js:155:20)
  at handler(node_modules/regenerator-runtime/runtime.js:167:13)
  at fn(node_modules/core-js/modules/es.promise.js:119:22)
  at fn(node_modules/core-js/internals/microtask.js:28:9)
  at tryCallOne(node_modules/react-native/node_modules/promise/setimmediate/core.js:37:12)
  at apply(node_modules/react-native/node_modules/promise/setimmediate/core.js:123:15)
  at callback(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:248:18)
  at _callTimer(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:112:7)
  at _callReactNativeMicrotasksPass(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:166:5)
  at _reactNativeMicrotasksCallback(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:418:12)
  at __callReactNativeMicrotasks(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:393:12)
  at fn(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:135:12)
  at __guard(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:370:9)
  at value(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:134:10)
  at value([native code])
  at value([native code])```

```Error: transceive fail
  at Error([native code])
  at _construct(node_modules/@babel/runtime/helpers/construct.js:12:53)
  at _construct(node_modules/@babel/runtime/helpers/construct.js:19:21)
  at Wrapper(node_modules/@babel/runtime/helpers/wrapNativeSuper.js:26:14)
  at NfcErrorBase(node_modules/react-native-nfc-manager/src/NfcError.js:1:1)
  at NfcErrorBase(node_modules/react-native-nfc-manager/src/NfcError.js:3:14)
  at buildNfcExceptionAndroid(node_modules/react-native-nfc-manager/src/NfcError.js:142:14)
  at call(node_modules/react-native-nfc-manager/src/NfcError.js:158:17)
  at tryCatch(node_modules/regenerator-runtime/runtime.js:63:40)
  at call(node_modules/regenerator-runtime/runtime.js:294:22)
  at tryCatch(node_modules/regenerator-runtime/runtime.js:63:40)
  at invoke(node_modules/regenerator-runtime/runtime.js:155:20)
  at handler(node_modules/regenerator-runtime/runtime.js:167:13)
  at fn(node_modules/core-js/modules/es.promise.js:119:22)
  at fn(node_modules/core-js/internals/microtask.js:28:9)
  at tryCallOne(node_modules/react-native/node_modules/promise/setimmediate/core.js:37:12)
  at apply(node_modules/react-native/node_modules/promise/setimmediate/core.js:123:15)
  at callback(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:248:18)
  at _callTimer(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:112:7)
  at _callReactNativeMicrotasksPass(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:166:5)
  at _reactNativeMicrotasksCallback(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:418:12)
  at __callReactNativeMicrotasks(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:393:12)
  at fn(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:135:12)
  at __guard(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:370:9)
  at value(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:134:10)
  at value([native code])
  at value([native code])```

```Error: no tech request available
  at Error([native code])
  at _construct(node_modules/@babel/runtime/helpers/construct.js:12:53)
  at _construct(node_modules/@babel/runtime/helpers/construct.js:19:21)
  at Wrapper(node_modules/@babel/runtime/helpers/wrapNativeSuper.js:26:14)
  at NfcErrorBase(node_modules/react-native-nfc-manager/src/NfcError.js:1:1)
  at NfcErrorBase(node_modules/react-native-nfc-manager/src/NfcError.js:3:14)
  at buildNfcExceptionAndroid(node_modules/react-native-nfc-manager/src/NfcError.js:142:14)
  at call(node_modules/react-native-nfc-manager/src/NfcError.js:158:17)
  at tryCatch(node_modules/regenerator-runtime/runtime.js:63:40)
  at call(node_modules/regenerator-runtime/runtime.js:294:22)
  at tryCatch(node_modules/regenerator-runtime/runtime.js:63:40)
  at invoke(node_modules/regenerator-runtime/runtime.js:155:20)
  at handler(node_modules/regenerator-runtime/runtime.js:167:13)
  at fn(node_modules/core-js/modules/es.promise.js:119:22)
  at fn(node_modules/core-js/internals/microtask.js:28:9)
  at tryCallOne(node_modules/react-native/node_modules/promise/setimmediate/core.js:37:12)
  at apply(node_modules/react-native/node_modules/promise/setimmediate/core.js:123:15)
  at callback(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:248:18)
  at _callTimer(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:112:7)
  at _callReactNativeMicrotasksPass(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:166:5)
  at _reactNativeMicrotasksCallback(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:418:12)
  at __callReactNativeMicrotasks(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:393:12)
  at fn(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:135:12)
  at __guard(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:370:9)
  at value(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:134:10)
  at value([native code])
  at value([native code])```
  
  
  app/android/build.gradle
  ```
buildscript {
    ext {
        buildToolsVersion = "30.0.2"
        minSdkVersion = 21
        compileSdkVersion = 31
        targetSdkVersion = 31
        ndkVersion = "21.4.7075529"

        // react-native-device-info
        supportLibVersion = "1.0.2"
        mediaCompatVersion = "1.0.1"
        supportV4Version = "1.0.0"
        googlePlayServicesIidVersion="17.0.0"

    }
```

package.json
```
    "react": "17.0.2",
    "react-native": "0.66.3",    
    "react-native-nfc-manager": "3.11.5",
```

Initially we had nfc-manager 3.11.0 out so not sure if these stacktraces happened in that version or the latest, but it's definedly still not working

**Is there some known issues with Android 12 at the moment? Is there something we can do to get around these?**

We are losing a bit of money because of this, so a bit urgent at least from our side 😄

whitedogg13, 2022-01-15:
Hi @Waltari10 sorry but I don’t get any Android 12 device for now, and I didn’t find any strange behavior by observing the native logs you provide.

BTW, I do get a Samsung Note 10, and there’s news said the Android 12 system upgrade is coming soon, I will do some tests after it arrives.

Waltari10, 2022-01-15:
So far issues with:

Pixel 6 Pro
Samsung s21 ultra
Google Pixel 6 Pro
Google Pixel 3
Google Pixel 4

Waltari10, 2022-01-15:
Thanks, @whitedogg13 I'd appreciate it! It actually seems like it's about fifty fifty whether it works on OS 12.

Also not quite sure yet whether the problem is in our own codebase or not. Def seems like Android OS 12 thing though.

Waltari10, 2022-01-18:
Kinda unrelated, but when I'm trying to debug this issue, and I open react-native-nfc-manager/android in Android Studio (Arctic Fox) none of the code completions/suggestions work. Any idea how to get those working?

whitedogg13, 2022-01-30:
Hi @Waltari10 I just got Android 12 for my Note 10+, and it works correctly with this library.

Can you try to run our demo app on your Android 12 devices to see if there's any error? 
- [play store link](https://play.google.com/store/apps/details?id=com.washow.nfcopenrewriter)
- [repo](https://github.com/revtel/react-native-nfc-rewriter)

Waltari10, 2022-02-17:
I think we found the issue. Not sure if this applies to other use cases, but in our case the NFC flags were somehow incompatible with certain devices.

It was working with 99% devices we tried. Then we tried with a Pixel 6 and a Samsung Flip phone, and we were unable to get the tag discovered event to fire.

Pixel 6 at least hasn't officially been released in Europe so it might be something with US device chipsets only. (no clue, but seemed to work on all EU phones we tried).

So first we tried with only FLAG_READER_NFC_V flag and it worked 99% of cases (EU phones):

```
          adapter.enableReaderMode(getCurrentActivity(),
                    new NfcAdapter.ReaderCallback() {
                        @Override
                        public void onTagDiscovered(final Tag tag) {
                            System.out.println("Tag found!");
                        }
                    },
                    NfcAdapter.FLAG_READER_NFC_V,
                    null);
```

Then we tried with the following flags and it started working with Pixel 6 and Samsung Flip (Same NFC tag being read):
  ```
        adapter.enableReaderMode(getCurrentActivity(),
                    new NfcAdapter.ReaderCallback() {
                        @Override
                        public void onTagDiscovered(final Tag tag) {
                            System.out.println("Tag found!");
                        }
                    },
                    NfcAdapter.STATE_ON |
                            NfcAdapter.STATE_TURNING_OFF |
                            NfcAdapter.FLAG_READER_NFC_V |
                            NfcAdapter.FLAG_READER_NFC_BARCODE |
                            NfcAdapter.FLAG_READER_SKIP_NDEF_CHECK |
                            NfcAdapter.FLAG_READER_NO_PLATFORM_SOUNDS,
                    null);
```

The integer values of all those flags is: 3 + 4+ 8 + 16 + 128 + 256 = 415 or 0b110011111 in binary.

The sensor we are reading is definedly NFCV sensor, but for some reason with Pixel 6 phone it didn't detect unless we had all those other extra flags.

So maybe this is a weird NFC tag manufacturer specific thing. Certainly some weirdness going on with different phone chipsets depending on continent...

We ended up creating a NFC solution from scratch on Android side. On iOS we have had none of these problems, so happily using this library there still.

So not sure if you can control those Flags from Javascript side with this library? It seems at least in some very specific corner cases it could be useful.

whitedogg13, 2022-02-19:
@Waltari10 Thanks for sharing this useful insight.

The `enableReaderMode` API you mentioned is indeed missing during V2 to V3 rewriting, so it's not there for now. I will bring it back later!

whitedogg13, 2022-02-19:
Hi @Waltari10 please use the latest `v3.13.2`, which fixes the `readerMode` issue. You can use it like this:

```javascript
import NfcManager, {
  NfcTech,
  NfcAdapter,
} from 'react-native-nfc-manager';
...
await NfcManager.requestTechnology([NfcTech.Ndef], {
  isReaderModeEnabled: true,
  readerModeFlags:
    // eslint-disable-next-line no-bitwise
    NfcAdapter.FLAG_READER_NFC_A |
    NfcAdapter.FLAG_READER_NFC_V |
    NfcAdapter.FLAG_READER_NFC_BARCODE |
    NfcAdapter.FLAG_READER_SKIP_NDEF_CHECK |
    NfcAdapter.FLAG_READER_NO_PLATFORM_SOUNDS,
});
```

The fix is in #496 

Hope that helps!

marcokoivisto, 2022-03-04:
I've tested out your example @whitedogg13 But unfortunately, I'm still encountering "transceive fail"  while transceiving data with the nfcAHandler :/ (Samsung S20FE, Android OS 12)

nick0504k, 2022-05-25:
public class ESEWrapper extends ReactContextBaseJavaModule implements LifecycleEventListener {
    private static final String TAG = "ESEWrapper";
    private final ReactContext reactContext;
    OfflineCbdcWalletApi api = new OfflineCbdcWalletApi();

    public ESEWrapper(ReactApplicationContext reactContext) {
        super(reactContext);
        reactContext.addLifecycleEventListener(this);
        this.reactContext = reactContext;
    }

    @Override
    public void onHostResume() {
        boolean result = this.api.beginNfcForegroundDispatch(getCurrentActivity());
        Log.d(TAG, "onHostResume beginNfcForegroundDispatch ==> " + result);
    }

    @Override
    public void onHostPause() {
        Log.d(TAG, "onHostPause");
    }

    @Override
    public void onHostDestroy() {
        boolean result = this.api.endNfcForegroundDispatch(getCurrentActivity());
        Log.d(TAG, "onHostDestroy endNfcForegroundDispatch ==> " + result);
    }
}

it works for me

closed 2022-09-09 by github-actions[bot]

AftabUfaq, 2025-06-22:
> * play store link

not avaliable in Pakistan