revtel/react-native-nfc-manager #481: [feat] sendRequestWithFlag for iOS 14+

whitedogg13, 2022-01-15:
as mentioned in #230 , we should bridge this API to JS

raydenvoldeskine, 2022-01-16:
While I believe this is generally a good idea, still the Apple docs are missing completely. I tried to use sendRequest() locally instead of sendCustomCommand and id did not work either for my case. Asked Apple Forums to explain behaviour: https://developer.apple.com/forums/thread/698545

raydenvoldeskine, 2022-01-17:
Finally I figured out my problem was still in message composition. I confirm I can achieve what I need with sendRequest() method.  This is (rather trivial) objective c extension: 

```
RCT_EXPORT_METHOD(iso15693_sendRequest:(NSDictionary *)options callback:(nonnull RCTResponseSenderBlock)callback)
{
    if (@available(iOS 14.0, *)) {
        if (!tagSession || !tagSession.connectedTag) {
            callback(@[@"Not connected", [NSNull null]]);
            return;
        }
        
        id<NFCISO15693Tag> tag = [tagSession.connectedTag asNFCISO15693Tag];
        if (!tag) {
            callback(@[@"incorrect tag type", [NSNull null]]);
            return;
        }

        RequestFlag flags = [[options objectForKey:@"flags"] unsignedIntValue];
        NSInteger commandCode = [[options objectForKey:@"commandCode"] integerValue];
        NSData *data = [self arrayToData:[options mutableArrayValueForKey:@"data"]];
        
        // https://developer.apple.com/documentation/corenfc/nfciso15693tag/3551933-sendrequestwithflag?language=objc
        
        [tag sendRequestWithFlag:flags commandCode:commandCode data:data completionHandler:^(NFCISO15693ResponseFlag responseFlag, NSData * _Nullable_result resp, NSError * _Nullable error) {
            if (error) {
                callback(@[getErrorMessage(error), [NSNull null]]);
                return;
            }
            
            callback(@[[NSNull null], [self dataToArray:resp]]);
        }];
        
    } else {
        callback(@[@"Not supported in this device", [NSNull null]]);
    }
}
```

whitedogg13, 2022-01-20:
Hi @raydenvoldeskine I have created PR #486 regarding this feature.  Please take a look and try it out, thanks!

s00se, 2022-01-27:
Does #486 also address #458 ?

whitedogg13, 2022-01-27:
Yes, that's what er're trying to do.

closed 2022-04-27 by whitedogg13

flyskywhy, 2026-03-02:
System NFC alert modal can't show on my Android phone, so be this wrapper [react-native-nfc-rw](https://github.com/flyskywhy/react-native-nfc-rw?tab=readme-ov-file) to read and write NfcV on iOS and Android , maybe you can ref to it.