revtel/react-native-nfc-manager #482: error: cannot find symbol flag = PendingIntent.FLAG_MUTABLE

queltalent, 2022-01-15:
Hi,
I am trying to build my project with "react-native-nfc-manager": "3.11.5", and I am getting this error:
node_modules/react-native-nfc-manager/android/src/main/java/community/revteltech/nfc/NfcManager.java:1080: error: cannot find symbol
          flag = PendingIntent.FLAG_MUTABLE;
                              ^
  symbol:   variable FLAG_MUTABLE
  location: class android.app.PendingIntent

My setup:

"react-native": "0.66.0",
"react-native-nfc-manager": "3.11.5",

buildToolsVersion = "30.0.2"
minSdkVersion = 21
compileSdkVersion = 31
targetSdkVersion = 31

mrmuke, 2022-01-16:
Same Problem.

queltalent, 2022-01-16:
Upgrade gradle tools sdk to 11 fixes this, close this issue.

closed 2022-01-16 by queltalent

ZanaNabaz, 2022-01-27:
> gradle tools sdk to 11 f

Gradle tools newest version is 7.2 ?

JdaieLin, 2022-02-05:
NfcManager.java compares the buildSDK version with Build.VERSION_CODES.M (which is 23)

```
  int flag = 0;
  if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
    flag = PendingIntent.FLAG_MUTABLE;
  }
```

However FLAG_MUTABLE is not added until API 31:
https://developer.android.com/reference/android/app/PendingIntent#FLAG_MUTABLE

I don't understand.

JdaieLin, 2022-02-05:
> NfcManager.java compares the buildSDK version with Build.VERSION_CODES.M (which is 23)
>
> ```
>   int flag = 0;
>   if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
>     flag = [...]

Just find the reason here: https://github.com/revtel/react-native-nfc-manager#android-12
Change the compileSdkTaget to 31 will fix it.

josiext, 2024-02-09:
> Just find the reason here: https://github.com/revtel/react-native-nfc-manager#android-12 Change the compileSdkTaget to 31 will fix it.

This works 🔥